anikau31/systemc-clang #8: LLVM_BUILD_DIR vs LLVM_INSTALL_DIR

mayagokhale, 2019-08-29:
Latest CMakelists.txt uses LLVM_INSTALL_DIR rather than LLVM_BUILD_DIR but paths.sh and paths.fish in scripts have not been updated to set LLVM_INSTALL_DIR.

rseac, 2019-08-30:
This should be fixed.  Ideally, we want cmake to find the LLVM package by itself; thus, not requiring us to specify the installation location for it.

mayagokhale, 2019-08-30:
That could be hard to do automatically if people have multiple LLVM versions installed.

rseac, 2019-10-04:
Commit https://github.com/anikau31/systemc-clang/commit/8af43bc0caf737d49a58a1c900325638bf27d8d8 completes this enhancement.  

The default behaviour is to find the Clang package if `LLVM_INSTALL_DIR` is not specified.  This variable can be specified either as an environment variable or via the `-DLLVM_INSTALL_DIR=` when invoking `cmake`.

closed 2019-10-04 by rseac