anikau31/systemc-clang #9: Othervars no longer being saved in FindPort and FindModuleDecl

mayagokhale, 2019-08-29:
In the master branch, the code to save off variables declared in a module has been removed. This results in an sc_buffer not being included in a module. In the example below, we lose        sc_buffer<double> i[ORDER]:

 SC_MODULE(FIR) {
        sc_in_clk clk;
        sc_in<double> sample;       
        sc_out<double> result;
        SC_CTOR(FIR) {      
                SC_METHOD(behavior);
                sensitive << clk.neg();
        }
private:
        static const int ORDER = 5;
        static const double c[ORDER+1];
        sc_buffer<double> i[ORDER];
        void behavior() {
                double sum = c[0] * sample.read();
                for (int j=1; j<=ORDER; ++j)
                        sum = sum + c[j] * i[j-1].read();
                result.write(sum);

                i[0].write(sample.read());
                for (int j=1; j<ORDER; ++j)
                        i[j].write(i[j-1].read());
        }
};

rseac, 2019-08-30:
Variables of a module as far as I recall are not identified unless they are ports or signals.  `sc_buffer` should not be recognized because the code explicitly looked for `sc_signal`.  Given that `sc_buffer` inherits from `sc_signal`, a general solution to this would be to rework `FindSignals.cpp` to identify any instance that has a base class of `sc_signal`.  Currently, I'm opening up a separate issue #10 for this.

A quicker way forward involves treating  `sc_buffer` as `sc_signal`.

mayagokhale, 2019-08-30:
In the previous version of systemc-clang, there was a variable _othervars to keep track of any other declarations that might occur in a module, and the sc_buffer was put into that list. That worked well since the declaration could be passed to the HDL generator. Maybe we could just restore that variable and associated methods.

rseac, 2019-08-31:
@mayagokhale Thank you.  I hadn't purposely removed this code.  It just seems that I had never merged it in to master.  Nonetheless, I've re-introduced it.  Please let me know if this fixes it.

closed 2019-09-04 by rseac