anikau31/systemc-clang #12: Errors compiling example plugin

gsp2, 2019-10-25:
Hello,

I am new to systemc-clang, and I am interested in trying to use systemc-clang to transform code. I tried to follow the instructions inside doc/plugins.md to write a plugin for systemc-clang. I took the example program that was listed in the first half of the document and saved it as "ModelDump.h" under src/. Then I combined the existing driver-tooling.cpp file with what was listed in the second half of instructions so that my final driver-tooling.cpp file looked like this:

#include "SystemCClang.h"
#include "PluginAction.h"
using namespace scpar;
using namespace std;

// Include your class
#include "ModelDump.h"

int main(int argc, const char **argv) {
  PluginAction<SystemCClang> systemc_clang(argc, argv);
  PluginAction<ModelDump> pa(argc, argv);
  return 0;
}

Then I compiled systemc-clang normally. After I ran make, after several warnings, there were 2 errors shown below (only the output starting from the 1st error is shown):

In file included from /data/loewenheim/a/gpajela/Downloads/systemc-clang/driver-tooling.cpp:7:
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/ModelDump.h:11:6: error: 
      no matching constructor for initialization of 'scpar::SystemCConsumer'
   : SystemCConsumer(ci) {
     ^               ~~
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/SystemCClang.h:61:9: note: 
      candidate constructor (the implicit copy constructor) not viable: no known
      conversion from 'clang::CompilerInstance' to
      'const scpar::SystemCConsumer' for 1st argument
  class SystemCConsumer : public ASTConsumer,
        ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/SystemCClang.h:69:7: note: 
      candidate constructor not viable: requires 2 arguments, but 1 was provided
      SystemCConsumer( CompilerInstance &, std::string top );
      ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/SystemCClang.h:110:59: error: 
      no matching constructor for initialization of 'ModelDump'
          return std::unique_ptr<clang::ASTConsumer>( new A(ci, top_) );
                                                          ^ ~~~~~~~~
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/SystemCClang.h:103:9: note: 
      in instantiation of member function
      'scpar::LightsCameraAction<ModelDump>::CreateASTConsumer' requested here
        LightsCameraAction( std::string topModule ):top_{topModule} { };
        ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/PluginAction.h:33:18: note: 
      in instantiation of member function
      'scpar::LightsCameraAction<ModelDump>::LightsCameraAction' requested here
      return new PluginConsumer(top_); }
                 ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/PluginAction.h:31:14: note: 
      in instantiation of member function
      'SimpleFrontendActionFactory<scpar::LightsCameraAction<ModelDump>
      >::create' requested here
    explicit SimpleFrontendActionFactory( std::string topModule ): top_{...
             ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/PluginAction.h:53:26: note: 
      in instantiation of member function
      'SimpleFrontendActionFactory<scpar::LightsCameraAction<ModelDump>
      >::SimpleFrontendActionFactory' requested here
      factory.reset( new SimpleFrontendActionFactory< LightsCameraAction...
                         ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/driver-tooling.cpp:11:27: note: 
      in instantiation of member function
      'PluginAction<ModelDump>::PluginAction' requested here
  PluginAction<ModelDump> pa(argc, argv);
                          ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/ModelDump.h:10:2: note: 
      candidate constructor not viable: requires single argument 'ci', but 2
      arguments were provided
 ModelDump( CompilerInstance& ci)
 ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/ModelDump.h:6:7: note: 
      candidate constructor (the implicit copy constructor) not viable: requires
      1 argument, but 2 were provided
class ModelDump
      ^
/data/loewenheim/a/gpajela/Downloads/systemc-clang/src/ModelDump.h:6:7: note: 
      candidate constructor (the implicit move constructor) not viable: requires
      1 argument, but 2 were provided
14 warnings and 2 errors generated.
CMakeFiles/systemc-clang.dir/build.make:62: recipe for target 'CMakeFiles/systemc-clang.dir/driver-tooling.cpp.o' failed
make[2]: *** [CMakeFiles/systemc-clang.dir/driver-tooling.cpp.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/systemc-clang.dir/all' failed
make[1]: *** [CMakeFiles/systemc-clang.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

Here are the versions of my software:
- systemc-clang: commit c8ee835
- CXX compiler: clang version 7.0.0-3~ubuntu0.18.04.1
- SystemC: 2.3.3
- OS: Ubuntu 18.04.3 LTS
- Platform: Linux x86_64

rseac, 2019-10-26:
Can you try the following?

1. Pull from master.  Some changes were brought back in that may help fix part of your issue.
2. Your plugin code should be something like the following.

```c++
#include "SystemCClang.h"
#include "PluginAction.h"
using namespace scpar;
using namespace std;

#include "ModelDump.h"

int main(int argc, const char **argv) {
  PluginAction<ModelDump> systemc_clang(argc, argv);
  return 0;
}
```
Notice the template argument to `PluginAction`.  
3. The constructor for SystemCConsumer changed a little.  Please see the updated `plugins.md` with the updated `ModelDump.h`.

gsp2, 2019-10-26:
I tried what you said, and it's working for me now. Thanks!

closed 2019-10-26 by gsp2