anikau31/systemc-clang #15: catch2 on mac

mayagokhale, 2019-10-25:
Now, there are additional errors being generated from the catch header when compiling on a mac with clang 11.0.0. The prior one was "raise" on line 10630 being undeclared. The new errors after the last push are line 10639 signalstack being undeclared, 10645, 10657, 10660 sigaction having the wrong number of arguments (3, when 1 is required).

rseac, 2019-10-26:
@zhuanhao-wu Can you investigate this?

zhuanhao-wu, 2019-10-26:
@mayagokhale Can you confirm the following:

The compiler is trying to find `raise`, `signalstack` etc from this header: `/usr/include/sys/signal.h`.
That is, is the error something similar to the following? (The line number could be slightly different)
```
In file included from main.cpp:2:
./catch.hpp:10696:9: error: use of undeclared identifier 'raise'
        raise( sig );
        ^
./catch.hpp:10705:9: error: use of undeclared identifier 'sigaltstack'
        sigaltstack(&sigStack, &oldSigStack);
        ^
./catch.hpp:10711:13: error: no matching constructor for initialization of 'sigaction'
            sigaction(signalDefs[i].id, &sa, &oldSigActions[i]);
            ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 3 were provided
struct  sigaction {
        ^
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 3 were provided
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit default constructor) not viable: requires 0 arguments, but 3 were provided
In file included from main.cpp:2:
./catch.hpp:10723:17: error: no matching constructor for initialization of 'sigaction'
                sigaction(signalDefs[i].id, &oldSigActions[i], nullptr);
                ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 3 were provided
struct  sigaction {
        ^
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 3 were provided
/usr/include/sys/signal.h:286:9: note: candidate constructor (the implicit default constructor) not viable: requires 0 arguments, but 3 were provided
In file included from main.cpp:2:
./catch.hpp:10726:13: error: use of undeclared identifier 'sigaltstack'
            sigaltstack(&oldSigStack, nullptr);
            ^
5 errors generated.
```

---

If that's the case, could you try the following?

It looks like the errors are due to missing headers, I installed the headers with the following command:

`open /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_{Your OS Version}.pkg`

On my machine, it's:
`open /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_10.14.pkg`

After that, simply recompile the project.

mayagokhale, 2019-10-26:
Those are the errors and I did as you suggested. However, the package installer reports an error that the package is incompatible with this version of macos. I am on 10.14.6 which is supposed to the the latest. I also updated xcode so it is uptodate and tried again but it reports the same problem. I will work on it some more.

Maybe the default is to turn tests off, because it has been building cleanly recently.

mayagokhale, 2019-10-26:
After I did the xcode update, I removed the entire build directory and started all over with cmake turning on xlat and enable_tests. It now builds catch2 and catch.

Scanning dependencies of target catch-test
[ 96%] Building CXX object tests/CMakeFiles/catch-test.dir/main.cpp.o
[ 98%] Building CXX object tests/CMakeFiles/catch-test.dir/catch-test.cpp.o
[100%] Linking CXX executable catch-test
[100%] Built target catch-test

closed 2019-10-26 by mayagokhale