anikau31/systemc-clang #16: Xlat code is not being run

mayagokhale, 2019-10-25:
# Description
The synthesis modules in the xlat plugin are not being run

# Reproducing the bug 

## Versions 
- systemc-clang: latest in synthesis branch
- CXX compiler: clang 11.0.0
- CXX flags: [i.e. llvm-config --cxxflags]
- SystemC:  2.3.3
- Operating System: macos

## Minimal Working Example (MWE)

When I use the run.sh script and run the filter example from the older systemc-clang revision, it doesn't go into the xlat module code. It does print XLAT plugin at the start, which verifies that it was compiled with the xlat plugin. When I did an nm of the systemc-clang executable, the methods in the xlat module don't appear. In my older version (sc-clang-xlat from my repo) there are lots of symbols related to xlat.

Also, I just redid the compilation, commenting out the automatic testing part since catch.hpp generates errors on the mac, and see that the files under plugins/xlat are not included in the compilation even though XLAT=on is printed out when it starts.

mayagokhale, 2019-10-25:
I think the problem is because SystemCClang.cpp doesn't call postFire() any more. I restored the previous postFire() code to the .cpp and .h, and the xlat module is now invoked in my local version.

mayagokhale, 2019-10-26:
index 6971560..5c6ec2a 100644
--- a/src/SystemCClang.cpp
+++ b/src/SystemCClang.cpp
@@ -3,6 +3,8 @@ using namespace scpar;
 using namespace clang;
 using namespace std;
 
+bool SystemCConsumer::postFire() { return true; }
+
 bool SystemCConsumer::fire() {
 
   os_ << "Top module: " << getTopModule() << "\n";
@@ -221,6 +223,7 @@ void SystemCConsumer::HandleTranslationUnit(ASTContext &context) {
   // ///////////////////////////////////////////////////////////////
 
   fire(); 
+  postFire();
 }
 
 SystemCConsumer::SystemCConsumer( CompilerInstance &ci, std::string top )

--------------

git diff SystemCClang.h
diff --git a/src/SystemCClang.h b/src/SystemCClang.h
index cd27400..ad8ae74 100644
--- a/src/SystemCClang.h
+++ b/src/SystemCClang.h
@@ -76,6 +76,7 @@ namespace scpar {
 
       // Virtual methods that plugins may override.
       virtual bool fire();
+      virtual bool postFire();
 
       virtual void HandleTranslationUnit(ASTContext &context);

rseac, 2019-10-26:
d6d329e  brings back the `prefire`, `postfire` methods into synthesis branch. Please let me know if this resolves it.

mayagokhale, 2019-10-26:
Yes, I did a pull and rebuilt and it resolved the problem. The xlat module is being called.

closed 2019-10-26 by mayagokhale