anikau31/systemc-clang #18: updates to synth branch

mayagokhale, 2019-11-01:
I am not a contributor on this project, so can't do commits. Here are some the updates that
- restore othervars being available from ModuleDecl. It used to be there and then disappeared. 
- updates to the xlat plugin to process othervars
- updates to the xlat plugin to process sensitivity list.
[diffs.txt](https://github.com/anikau31/systemc-clang/files/3799700/diffs.txt)

rseac, 2019-11-05:
@mayagokhale One way to get updates would be to open up a pull request.  I believe this can be done for the synthesis branch as well.

Alternatively, you've been added as a contributor.

mayagokhale, 2019-11-05:
Thank you. I forked the systemc-clang repo and am making updates to my fork. I will put in pull requests from the fork. I accepted the contributor invitation, but for now will work in the fork.

closed 2019-11-05 by mayagokhale