anikau31/systemc-clang #28: Multiple instances/declarations when matching templated module using ASTMatcher

zhuanhao-wu, 2019-12-10:
# Description

When matching the templated modules, such as the sreg module in the example, it might create multiple module declarations for multiple instantiations with different type parameters.

# Reproducing the bug 

Running the `sreg-test` should reveal the problem. 

# Expected Behavior
The number of instances/declarations should match the number of instantiations in the code.

# Actual Behavior
If we look at the number of instances/declarations generated for the three instances with different type parameters, we can observe there are 9 in total.

zhuanhao-wu, 2019-12-10:
b69e745
Added test for the issue.

rseac, 2019-12-23:
@zhuanhao-wu I've pushed in a new branch called `integrate-matchers` at c493575f11b5b98377fc87162cdcdd8fe4e33c2e.  Can you ensure that the sreg test, which is passing for templated modules, is indeed correct.

rseac, 2020-01-28:
Fixed in #35.

closed 2020-01-28 by rseac