anikau31/systemc-clang #44: TypedefType not getting processed

mayagokhale, 2020-02-13:
In the following tree, the FindTemplateTypes *te->enumerate() call is not enumerating the type arguments. First the source code, then the tree. In XlatEntryMethod.cpp where the enumerate call is occuring, there are no arguments, which generates a variable declaraion of notype.

(in zfp.h, line 231)
expo_t emax = s_ex.data_r();

Tree reported by the dump routine:
ProcessVarDecl var name is emax
ProcessVarDecl type name is fwd_cast<struct fp_t<11, 52>, 2>::expo_t
TypedefType 0x5748e20 'fwd_cast<struct fp_t<11, 52>, 2>::expo_t' sugar
|-Typedef 0x5748c68 'expo_t'
`-ElaboratedType 0x5748c10 'typename struct fp_t<11, 52>::expo_t' sugar
  `-TypedefType 0x562a820 'fp_t<11, 52>::expo_t' sugar
    |-Typedef 0x5619348 'expo_t'
    `-TemplateSpecializationType 0x56192d0 'sc_uint<11>' sugar sc_uint
      |-TemplateArgument expr
      | `-SubstNonTypeTemplateParmExpr 0x5619198 'int'
      |   `-IntegerLiteral 0x5619178 'int' 11
      `-RecordType 0x56192b0 'class sc_dt::sc_uint<11>'
        `-ClassTemplateSpecialization 0x56191c0 'sc_uint'

rseac, 2020-02-13:
This should be cleaned up for template instantiations.  In particular, when the template argument is a TypedefType, or a ClassTemplateSpecializationType, or a DependentType, or a CXXRecordType.   So, a first step is to enable visiting template instantiations in FindTemplateTypes, and move the code for specific template argument types as Visit*.
```
bool shouldVisitTemplateInstantiations() const { return true;}
```

rseac, 2020-02-15:
@mayagokhale Under the branch [`hotfix/44`,](https://github.com/anikau31/systemc-clang/tree/hotfix/44) I have rewritten `FindTemplateTypes` class.  This should handle Typedefs in addition to accessing the template arguments.  

The template arguments are stored in a `std::vector` of pairs of <string,Type*> where the string is the name of the template argument, and Type* is a pointer to the Type pointer to the node for that type. 

For example, `sc_signal<sc_uint<16>>` should end up having `sc_signal sc_uint 16` as the three strings in the `std::vector`.  

If you can confirm that this is okay, then I can merge these changes into master.

mayagokhale, 2020-02-16:
It looks great. For the emax declaration shown in the original comment, the type information is now being found, and the following is generated (which is correct):
hPortsigvarlist  NONAME [
            hVardecl emax [
              hTypeinfo  NONAME [
                hType sc_uint NOLIST
                hType 11 NOLIST
              ]

closed 2020-02-16 by mayagokhale