anikau31/systemc-clang #65: User-defined variables don't appear in list of othervars of an sc_module

mayagokhale, 2020-03-11:
Since sc_stream isn't synthesizable, in the llnl examples, we are using the alternate synthesizable version defined in sc_rvd.h. When sc_rvd types are used, other classes such as sreg define sc_stream to be aliased to sc_rvd:
sreg.h:template <typename T> using sc_stream_in = sc_rvd_in<T>;
sreg.h:template <typename T> using sc_stream_out = sc_rvd_out<T>;

However, when those types are used to declare variables at the top level of an sc_module, they don't appear in any of the lists. Since they are not input or output stream ports or signals, shouldn't the appear in OtherVars? I can't find s_fp, m_fp, or m_ex declarations anywhere.

If you look at find_emax:
SC_MODULE(find_emax)
{
        typedef typename FP::expo_t expo_t;

        /*-------- ports --------*/
        sc_in<bool> clk;
        sc_in<bool> reset;

        sc_stream_in <FP> s_fp;
        sc_stream_out<FP> m_fp;
        sc_stream_out<expo_t> m_ex;

        /*-------- registers --------*/
        sc_signal<sc_uint<2*DIM> > count;
        sc_signal<expo_t> emax;
        sc_signal<bool> emax_v;

        /*-------- channels --------*/
        sc_signal<bool> c_sync;
        sc_stream<FP> c_fp;
        sc_stream<expo_t> c_ex;

        /*-------- modules --------*/
        sfifo_cc<FP> u_que_fp;
        sreg<expo_t,FWD_REV> u_reg_ex;

The sc_stream declarations at the source level get redefined to sc_rvd types. In the AST dump output, those sc_rvd variables don't appear:

============== DUMP the MODULEDECL ======================
Module declaration name: find_emax
# Instances:
 none 
# Port Declaration:

Input ports: 2
Start printing ports
{
    "port_arguments": {
        "sc_in": [
            "_Bool"
        ]
    },
    "port_name": "clk"
}{
    "port_arguments": {
        "sc_in": [
            "_Bool"
        ]
    },
    "port_name": "reset"
}
Output ports: 0

Inout ports: 0
{
    "port_arguments": {
        "sc_uint": [
            "11"
        ],
        "sreg": [
            "1",
            "sc_uint"
        ]
    },
    "port_name": "u_reg_ex"
}Ports
{
    "clk": {
        "port_arguments": {
            "sc_in": [
                "_Bool"
            ]
        },
        "port_name": "clk"
    },
    "number_of_in_ports": 2,
    "reset": {
        "port_arguments": {
            "sc_in": [
                "_Bool"
            ]
        },
        "port_name": "reset"
    }
}
{
    "number_of_output_ports": 0
}
{
    "number_of_inout_ports": 0
}
{
    "number_of_instream_ports": 0
}
{
    "number_of_outstream_ports": 0
}{
    "number_of_other_vars": 1,
    "u_reg_ex": {
        "port_arguments": {
            "sc_uint": [
                "11"
            ],
            "sreg": [
                "1",
                "sc_uint"
            ]
        },
        "port_name": "u_reg_ex"
    }
}

# Signal Declaration:
Signals
{
    "c_sync": {
        "signal_arguments": [
            "_Bool"
        ],
        "signal_name": "c_sync",
        "signal_type": "sc_signal"
    },
    "count": {
        "signal_arguments": [
            "sc_uint",
            "4"
        ],
        "signal_name": "count",
        "signal_type": "sc_signal"
    },
    "emax": {
        "signal_arguments": [
            "sc_uint",
            "11"
        ],
        "signal_name": "emax",
        "signal_type": "sc_signal"
    },
    "emax_v": {
        "signal_arguments": [
            "_Bool"
        ],
        "signal_name": "emax_v",
        "signal_type": "sc_signal"
    },
    "number_of_signals": 4
}

mayagokhale, 2020-03-11:
I'm wondering if the AST matcher isn't finding it due to the "using" notation resulting in a different type of node.

rseac, 2020-03-11:
@mayagokhale This is most likely the reason.  I suspect that the way the AST is constructed when `using` is used that the type of every FieldDecl is an alias type.  Then, one must do another level of indirection to access the actual type.  

I will investigate this.

rseac, 2020-03-12:
I believe a good fix for this would be to change the way the port matchers work to use the desugared type.  

```c++
auto makePortHasNameMatcher(const std::string &name) {
return fieldDecl(hasType(hasUnqualifiedDesugaredType(recordType(
    hasDeclaration(cxxRecordDecl(hasName(name)))))));
}
```

rseac, 2020-03-12:
@mayagokhale This should be fixed.

rseac, 2020-03-19:
With the rework, the use of `using` for sc_stream will render `sc_rvd_*` the pointers to the respective AST nodes are accessible in the internal data structure that refers to streams instead of others.

closed 2020-03-19 by rseac