anikau31/systemc-clang #66: reset_signal_is not being recognized

mayagokhale, 2020-03-11:
In the latest master branch, the systemc construct reset_signal_is is not being matched. Earlier today before I pulled the latest version of master, it appeared as a sensitivity list variable.

rseac, 2020-03-11:
@mayagokhale  It is my understanding that `reset_signal_is` provides a mechanism to generate the reset logic for the given signal.  However, in earlier synthesis drafts, `reset_signal_is` was not supported.  In the most recent one, [v1.4](https://www.accellera.org/images/downloads/standards/systemc/SystemC_Synthesis_Subset_1_4_7.pdf), it suggests that for SC_METHOD, it is not supported.

If we wish to support it, then what should it be represented as?  It seems to be a function call that expands to generating reset logic ...

mayagokhale, 2020-03-11:
You're right. Looking at the spec more carefully, I see that it isn't supported. We'll revise the example to use a synthesizable replacement.

closed 2020-03-11 by mayagokhale