anikau31/systemc-clang #89: Replace SignalContainer with PortDecl in ModuleDecl

rseac, 2020-04-08:
# Description
The role of `PortDecl` is the same as `SignalContainer`.  `SignalContainer` could simply be replaced.

This will also fix some of the memory free issues noted in `ModuleDecl.cpp` in function `addPorts` where the translation from `PortDecl` is done to `SignalContainer`.  

Related to #88

mayagokhale, 2020-04-08:
This would help simplify code in the Xlat plugin. Currently much of the code to process port decls is duplicated in the Signal processing due to their being distinct classes. Having a single class where there is a field indicating  signal or port, and if port, the direction, would eliminate the duplication.

rseac, 2020-04-30:
At the moment, `SignalContainer` is derived from `Portdecl`.  Would that be a good way to remove the duplicated code for signal processing?

In doing this, I noted that `Signal` itself is just `SignalContainer`.  So, I could potentially have `Signal`  be derived from `PortDecl` (the name of this class would change to something more meaningful), and everything should still work ...

rseac, 2020-04-30:
Fixed in 592d78d1ceadc5ab8b29dfae8c448b93268a9747

closed 2020-04-30 by rseac