anikau31/systemc-clang #98: Parse array of ports for their netlist

rseac, 2020-04-30:
# Description
Parse array of ports appropriately, and build a netlist for it. 
The example that should pass is the `port-array-example.cpp`

rseac, 2020-08-09:
The parsing of array of ports and signals is being done so that its array size is being found. However, what the ports are bound to is not parsed yet.

rseac, 2020-08-17:
Fixed. 203ab4c

mayagokhale, 2020-08-18:
What value is returned for array size if is_array is false? In the code in PortDecl.cpp, there didn't seem to be an initial value set for it. is_array is initialized to false but the size isn't. It would be convenient if the size were zero if it isn't an array.

rseac, 2020-08-18:
I've initialized it to 0 now in PortDecl.

closed 2020-08-18 by mayagokhale