anikau31/systemc-clang #107: Error reporting

mayagokhale, 2020-05-14:
It would be good to have a standard error/warning reporting mechanism for the plugin. Maybe there is already something in place and I haven't found it. I'm looking for the best way to report to the user that "break" statements aren't handled and will be ignored.

rseac, 2020-05-15:
@mayagokhale You're correct.  We probably do need a good way to do this and be consistent in the main tool too. I wonder if this could guide us: [https://llvm.org/docs/ProgrammersManual.html#error-handling ](https://llvm.org/docs/ProgrammersManual.html#error-handling )

rseac, 2020-07-09:
I think a good way to do this would be to use `LLVM_DEBUG()` macro. It allows one to turn on and off debug output given certain types. I've added the code in xlat, and systemc-clang to allow for this. Essentially, with the use of `LLVM_DEBUG()`, one can now pass the `--debug` flag to the tool to print all the debug outputs. Alternatively, one could selectively output the debug output by using the `--debug-only` flag.  The way to set this would be to set `DEBUG_TYPE` in the respective header. 

Currently, examples that use this include the `SensitivityMatcher` and `InstanceMatcher` in `src/matchers/`.

rseac, 2020-07-09:
We should create a macro that wraps `LLVM_DEBUG()` within it. This makes it easy to use the corresponding code within our debug.
```c++
#undef DEBUG_TYPE
#define DEBUG_TYPE some_name
```

rseac, 2020-07-09:
There seems to be no need to do this. LLVM Debug allows for this via the ``DEBUG_WITH_TYPE()`` macro.

rseac, 2020-08-09:
I'm assuming that we are good with this approach.

closed 2020-08-09 by rseac

mayagokhale, 2020-08-10:
Yes, that will work well.