anikau31/systemc-clang #108: Submodules and their port bindings

mayagokhale, 2020-05-14:
The xlat plugin is not handling submodule instantiation. Submodules appear in the OtherVars list in a module and are being processed as regular variables, not as submodules to be instantiated. Processing the vector of submodules needs to be done, and the plugin must not include them as variables with flattened declarations. @mayagokhale needs to add this code.

rseac, 2020-05-15:
@mayagokhale One way to access the sub-modules is to access the following from a `ModuleDecl`: https://github.com/anikau31/systemc-clang/blob/master/src/model/ModuleDecl.h#L125
https://github.com/anikau31/systemc-clang/blob/master/src/model/ModuleDecl.h#L174 

This would provide access to the `ModuleDecl` for the corresponding sub-module instance.

mayagokhale, 2020-05-28:
Submodule processing has been implemented.

closed 2020-05-28 by mayagokhale