anikau31/systemc-clang #109: modulemap missing module?

mayagokhale, 2020-05-20:
I wrote a simplified testbench z1test.cpp that doesn't include the simulation code in zfpsynth/zfp1/test.cpp. 
In z1test.cpp, the sc_main calls a single module mymodule() and mymodule has a single component find_emax.
In the modulemap, there are two modules, find_emax and sreg. However, mymodule doesn't appear. I wasn't sure why. I'm trying to test the port bindings of submodules.
z1test.cpp is checked into the scratchllnl branch and also included here.

#define __SYNTHESIS__

#if defined(__SYNTHESIS__)
#define RVD
#endif

#if !defined(RLEVEL)
#define RLEVEL false
#endif

#include "../shared/pulse.h"
#include "../shared/zfp.h"

#define EXPOW 11
#define FRACW 52
// #define EXPOW  8
// #define FRACW 23
#ifndef DIMS
#define DIMS 1
#endif
#define BLOCK_CNT 2
#define BLOCK_LEN fpblk_sz(DIMS) /* values per block */

typedef fp_t<EXPOW,FRACW> fpn_t;
typedef fpn_t enc_t;
typedef typename fpn_t::expo_t expo_t;

SC_MODULE(mymodule)
{
  sc_clock clk; 
  sc_signal<bool> reset;
  sc_stream<fpn_t> c_tb_send_fp;
  sc_stream<enc_t> c_dut_enc;
  sc_stream<expo_t> c_dut_expo;
  find_emax<fpn_t, DIMS> u_dut;
  SC_CTOR(mymodule) : u_dut("u_dut")
  {
    // connect DUT
    u_dut.clk(clk);
    u_dut.reset(reset);
    u_dut.s_fp(c_tb_send_fp);
    u_dut.m_fp(c_dut_enc);
    u_dut.m_ex(c_dut_expo);
  }
};

#if DIMS > 2
#error "block array only supports 1 or 2 dimensions"
#endif

int sc_main(int argc , char *argv[])
{

  mymodule("mymodule");

}

mayagokhale, 2020-05-20:
Also, the find_emax module doesn't appear to have submodules. are submodules only collected for instances of modules and not the module itself? I changed the Xlat.cpp code to access the modules vector and not push into each instance since there's no point in generating hcode for identical instances.

rseac, 2020-05-20:
1. Could you try 
```c++
int sc_main() {
  mymodule mymod("mymod");
}
```
2. `find_emax` has at least the `sreg` as a submodule.  Submodules are collected of instances of modules.

mayagokhale, 2020-05-20:
Thank you. I made that change and now it is finding mymod. I also went back to checking the instances rather than the main module declaration. however, submodules are still not showing up. mymodule should show find_emax as a submodule and find_emax should show sreg.

Here is the only line I modified from the original file, as suggested:

mymodule mymod("mymod");

Modules: 
grep hModule z1test_hdl.txt
  hModule mymodule [
  hModule find_emax [
  hModule sreg [

submodules:

grep "submodule count" x
submodule count is 0
submodule count is 0
submodule count is 0

mayagokhale, 2020-05-20:
Here is debug output from one of the matchers:
=> var_type_name find_emax<fpn_t, 1> var_name u_dut
Inserting FD instance
 ===================== INSTANCE MATCHER ==================== 
=> var_name mymod var_type_name struct mymodule instance_name mymod
=> processing submodule
## InstanceArgumentMatcher
@@ ctor expr
=> submodule_instance_name u_dut parent name: mymodule
Inserting VD instance
-------

however find_emax doesn't show up as a submodule in the instance of mymod.

mayagokhale, 2020-05-20:
Looking more closely at the generated output, it seems that the u_dut from mymod is appearing in the variable list of mymod. I'm not sure how to detect from its type declaration that it is a module.

rseac, 2020-05-20:
@mayagokhale I added a test that used the driver you created. I've also added it into the repo.  The below is an example of accessing the `ModuleDecl` that holds the instance information for `u_dut`. 

https://github.com/anikau31/systemc-clang/blob/master/tests/xlat/find-emax.cpp#L39-L58

mayagokhale, 2020-05-21:
Thank you. In that example, find_emax is showing up as a submodule of mymodule. however, the only port bindings associated with mymodule are clk and reset. I also dumped the portbindings of find_emax and sreg, which are instantiated in mymodule, but only clk and reset are showing up in the port bindings.

rseac, 2020-05-21:
For `mymodule`, I see that all the `sc_stream` ports (`c_tb_send_fp`, `c_dut_enc`, `c_dut_expo`) appear under other vars. So, they do get recognized, but there does seem to be an issue in not categorizing them appropriately. 

For `find_emax`, I see ports in `instream_ports`, `outstream_ports`, and `other_vars`. I've pasted what I see below for the two modules.

I can confirm that the port bindings for the ports other than clk and reset are not being captured. Opened #111 for this.

**find_emax**
```
-- Instance name: u_dut
Module declaration name: find_emax
# Instances:
 none 
# Port Declaration:
(sc_in) sc_in (sc_in) _Bool (sc_in) sc_in (sc_in) _Bool {
    "sc_in": [
        "_Bool"
    ]
}(sc_in) sc_in (sc_in) _Bool (sc_in) sc_in (sc_in) _Bool {
    "sc_in": [
        "_Bool"
    ]
}(sc_rvd_in) sc_rvd_in (sc_rvd_in) fp_t (fp_t) 11 (fp_t) 52 (sc_rvd_in) sc_rvd_in (sc_rvd_in) fp_t (fp_t) 11 (fp_t) 52 {
    "fp_t": [
        "11",
        "52"
    ],
    "sc_rvd_in": [
        "fp_t"
    ]
}(sc_rvd_out) sc_rvd_out (sc_rvd_out) fp_t (fp_t) 11 (fp_t) 52 (sc_rvd_out) sc_rvd_out (sc_rvd_out) fp_t (fp_t) 11 (fp_t) 52 {
    "fp_t": [
        "11",
        "52"
    ],
    "sc_rvd_out": [
        "fp_t"
    ]
}(sc_rvd_out) sc_rvd_out (sc_rvd_out) sc_uint (sc_uint) 11 (sc_rvd_out) sc_rvd_out (sc_rvd_out) sc_uint (sc_uint) 11 {
    "sc_rvd_out": [
        "sc_uint"
    ],
    "sc_uint": [
        "11"
    ]
}(sc_rvd) sc_rvd (sc_rvd) fp_t (fp_t) 11 (fp_t) 52 (sc_rvd) sc_rvd (sc_rvd) fp_t (fp_t) 11 (fp_t) 52 {
    "fp_t": [
        "11",
        "52"
    ],
    "sc_rvd": [
        "fp_t"
    ]
}(sc_rvd) sc_rvd (sc_rvd) sc_uint (sc_uint) 11 (sc_rvd) sc_rvd (sc_rvd) sc_uint (sc_uint) 11 {
    "sc_rvd": [
        "sc_uint"
    ],
    "sc_uint": [
        "11"
    ]
}(rvfifo_cc) rvfifo_cc (rvfifo_cc) fp_t (fp_t) 11 (fp_t) 52 (rvfifo_cc) 3 (rvfifo_cc) 0 (rvfifo_cc) rvfifo_cc (rvfifo_cc) fp_t (fp_t) 11 (fp_t) 52 (rvfifo_cc) 3 (rvfifo_cc) 0 {
    "fp_t": [
        "11",
        "52"
    ],
    "rvfifo_cc": [
        "fp_t",
        "3",
        "0"
    ]
}(sreg) sreg (sreg) sc_uint (sc_uint) 11 (sreg) 1 (sreg) 0 (sreg) sreg (sreg) sc_uint (sc_uint) 11 (sreg) 1 (sreg) 0 {
[... 379 lines cut ...]
        ],
        "signal_name": "reset",
        "signal_type": "sc_signal"
    }
}

# Processes:
Processes
{
    "number_of_processes": 0
}
# Port binding:0
{
    "number_of_ports_bound": 0
}
# Signal binding:
 none
}{
    "instance_name": "NONE",
    "module_name": "mymodule",
    "nested_modules": [
        "u_dut"
    ]

```

rseac, 2020-05-29:
@mayagokhale  I think this should also be resolved in ee15dd503a1677ff9e9ad02c85d402cc240c4ebf

rseac, 2020-06-05:
Fixed.

closed 2020-06-05 by rseac