anikau31/systemc-clang #114: typedef to SC_MODULE not being found as submodules

mayagokhale, 2020-05-28:
Submodules and port bindings are being provided to the xlat plugin, and hcode is being generated. (user defined types are not yet completely implemented in xlat plugin).

However, it appears that a submodule that is indirectly invoked via a typedef is not being matched by the front end and is not being added to the submodule list. The example is in z1test.cpp in the zfp1 directory, which is a much simplified version of test.cpp. It instantiates find_emax, which through sv_rvd.h includes rvfifo_cc.h, which has the following declaration:

fifo_t u_fifo;

above in that file, fifo_t is typedef'ed to be fifo_cc with some template parameters:
typedef fifo_cc<T,/*IW*/IW,/*FWFT*/true,/*RLEV*/RLEV,/*FLEV*/false> fifo_t;

In fifo_cc.h, fifo_cc is an SC_MODULE:
template <typename T, int IW, bool FWFT, bool RLEV, bool FLEV>
SC_MODULE(fifo_cc)

Perhaps due to the indirection through a typedef, the sc_module is not being matched.

rseac, 2020-05-28:
This should be recognized now.

rseac, 2020-06-05:
Fixed.

closed 2020-06-05 by rseac