anikau31/systemc-clang #123: port bindings list appears in instantiated module rather than instantiating module

mayagokhale, 2020-06-18:
Possibly I am accessing the module information incorrectly, but it seems as if the port binding list for an instantiated module is stored with the instantiated module rather than the parent instantiating module.
For example in z1test.cpp, the top level module mymodule instantiates a find_emax module and names it u_dut. The mymodule constructor establishes the port bindings between signals declared in mymodule and ports declared in find_emax. However, the port binding list is seen in an instance of find_emax, and therefore emitted in the hcode of the find_emax instance.

mayagokhale, 2020-06-19:
A possible method to work with this from the xlat plugin is in module processing: loop over the submodules and pull the port binding list from the submodule instance. I will work on this today in conjunction with the new way of traversing modules by starting with the top module and recursively instantiating submodules rather than iterating through the module list with an inner loop over each module instantiation.

rseac, 2020-06-19:
@mayagokhale  If we want to make some changes, we can certainly do that.

mayagokhale, 2020-06-19:
Let me try this way first. Since there is a unique module decl for each instantiation, there is no ambiguity in names, so as long as I can hoist the port binding list to the instantiating module in xlat, I can generate that list where Z wants it. I have to redo the way I'm generating modules anyway to attach a unique module instance name to the site where it is instantiated.

rseac, 2020-06-22:
@mayagokhale I'm assuing the PR enabled you to work with what we have?

mayagokhale, 2020-06-22:
I modified my code to examine the submodules and then hoist the portbindings up one level. Logically the port bindings belong up a level, but it all works so we can leave it this way. I’ll add a comment that they are stored at the lower level.

Maya

rseac, 2020-06-22:
I understand. I could move this bug as an enhancement, and fix it at a later date if it still makes sense.

mayagokhale, 2020-06-22:
that sounds good. it’s not holding anything up.

rseac, 2020-12-12:
Fixed in #428884c

closed 2020-12-12 by rseac