anikau31/systemc-clang #124: Modules and Module instantiations

mayagokhale, 2020-06-19:
I would like to understand what information is stored with module definitions and what is stored with module instantiations, and why.
In the current checked in version of the xlat plugin, the loop iterates over every instance of every module and generates a hModule for each instance. When there are multiple instantiations of a module, e.g. sreg, the entire module definition is repeated.
Today, I changed the main loop to iterate over each unique module definition and not over all its instantiations since there is no need to have the same module definition being repeated each time a module is instantiated.
When iterating over the module definitions, all the ports, signals, and variables are appearing in the module definition ModuleDecl. However, the submodules, port bindings, and processes are not.
I don't use the submodule list because submodules appear as other variables, but do need the port bindings and processes. Should those also be included in the module definition? Within the scope of the module definition, shouldn't they be the same for all instantiations? 
Then we would have a single top level module, and all the submodules referenced as variables in the top level module would have port bindings defined in the top level module. For the other modules, they would also define port bindings for modules instantiated in their definitions, and the port bindings would use the signal/port name mapping in the same way as is done for the top level.

rseac, 2020-06-19:
Module definitions (declarations) only carry partial information, as you discovered. Module instantiations carry all the information that are in the module definition and more that are specific to the instantiation. The reason to have this is that instances can have some information that is different. As a simple example, consider the instance names that are different in different instances of the same type. 

I would think that we should mainly be accessing the module instances, and not the definitions. The definitions are just a way to group the instances for that definition.

mayagokhale, 2020-06-19:
OK. I think rather than having a main loop over all the instances of a module as I was doing in the plugin, I should start with the top level module, generate its definition, and then recursively traverse its submodules, each of which will be a particular instance, and generate a unique definition for that submodule. For non-templated modules, that will cause a lot of redundancy, but for templated ones with types as template parameters, it has to be done. 
For our examples, it appears that the first module instantiated from sc_main is first in the module list, so for now, I can assume that is the top level module and start the traversal with it.

closed 2020-06-19 by mayagokhale