anikau31/systemc-clang #130: Sensitivity list not appearing in entryfunction container accessed from Module decl

mayagokhale, 2020-07-02:
The module decl stores a vector of entry function containers:
vector<EntryFunctionContainer *> getEntryFunctionContainer();
which is how the Xlat plugin has been accessing the the processes of a module and the sensitivity lists of the process. However, now, the sensitivity lists are empty.
In the matcher example for sensitivity lists, the lists are accessed through the ProcessMap. However, the same entry function vector in the process map should also be in the module decl, so it's not clear why the processes accessed from getEntryFunctionContainer don't have sensitivity lists.

Also, how do you tell if a module is an instance or the main definition?

mayagokhale, 2020-07-03:
I have revised the Xlat.cpp code to conform to the example shown. However, the sensitivity list is still not appearing. In this use case, there are 2 levels of nested modules before the module with processes that contain sensitivity lists. I wonder if the getNestedModuleDecl() is returning the declaration or instance. When I look at the dereferenced module decl pointer in lldb, it shows the instance name as "".
I have checked in the latest version of Xlat into scratchllnl. I'm not sure what to do next.

rseac, 2020-07-08:
Working.

closed 2020-07-08 by rseac