anikau31/systemc-clang #137: VarDecl pointer for const variable declaration in sc_module is not supplied in OtherVars

mayagokhale, 2020-08-07:
In fifo_cc.h from the llnl-examples/zfpsynth/shared directory, the following constructs appear:
	static constexpr int MAX_DEPTH = (1 << IW);
	const unsigned depth = MAX_DEPTH;
In the OtherVars list, both appear. There is a VarDecl pointer for MAX_DEPTH and it shows the initializer. However there isn't a VarDecl pointer for depth, and therefore an initializer from a VarDecl can't be found:

object name is depth and h_op is 22
// vardecl pointer is null, therefore the VarDecl isn't dumped

object name is MAX_DEPTH and h_op is 22
// vardecl pointer is valid, therefore the VarDecl is dumped and the initializer appears.
VarDecl 0x11268f7b8 </Users/gokhale2/src/systemc-clang-top/systemc-clang/tests/../examples/llnl-examples/zfpsynth/zfp1/../shared/fifo_cc.h:39:2, col:43> col:23 referenced MAX_DEPTH 'const int' static constexpr cinit
`-ParenExpr 0x11268f890 <col:35, col:43> 'int'
  `-BinaryOperator 0x11268f870 <col:36, col:41> 'int' '<<'
    |-IntegerLiteral 0x1104d06c0 <col:36> 'int' 1
    `-SubstNonTypeTemplateParmExpr 0x11268f850 <col:41> 'int'
      `-IntegerLiteral 0x11268f830 <col:41> 'int' 3

rseac, 2020-08-09:
A member is either of FieldDecl or VarDecl type. I'm storing them as one of its base class, Decl. For the `depth` variable it is a FieldDecl and not a VarDecl. So, you could use `getAsFieldDecl()` instead of `getAsVarDecl()`. Note that the I just pushed in a change to rename `getFieldDecl()` to `getAsFieldDecl()` (9d8ed2aa7a09fa81b3438d504a20a889cb7612b0)

mayagokhale, 2020-08-10:
I will change the code to look for either one.

closed 2020-08-10 by mayagokhale