anikau31/systemc-clang #147: constexpr function call as template specialization parameter

mayagokhale, 2020-09-08:
The template tree for the following declaration does not seem to resolve the constexpr function call to a constant. Ideally the template tree would show sc_bv<16>. Declaration and Tree follow.

Declaration
sc_signal<sc_bv<fpblk_sz(DIM)> > c_bplane[planes];

Debug info with tree:
object name is c_bplane
generatetype node name is sc_bv type follows
TemplateSpecializationType 0x7fce2ecf50a0 'sc_bv<fpblk_sz(2)>' sugar sc_bv
|-TemplateArgument expr
| `-ConstantExpr 0x7fce2ecf5078 'int' Int: 16
|   `-CallExpr 0x7fce2ecf5050 'int'
|     |-ImplicitCastExpr 0x7fce2ecf5038 'int (*)(int)' <FunctionToPointerDecay>
|     | `-DeclRefExpr 0x7fce2e82dad8 'int (int)' lvalue Function 0x7fce2e7c3740 'fpblk_sz' 'int (int)'
|     `-SubstNonTypeTemplateParmExpr 0x7fce2ecf5018 'int'
|       `-IntegerLiteral 0x7fce2ecf4ff8 'int' 2
`-RecordType 0x7fce2ece9d00 'class sc_dt::sc_bv<16>'
  `-ClassTemplateSpecialization 0x7fce2ece9c20 'sc_bv'

rseac, 2020-09-08:
The issue is in `FindTemplateTypes` where the `VisitDeclRefExpr` sees the fpblk_sz, but doesn't desugar it. The reason why it doesn't is because the `VisitDeclRefExpr` is used for user-defined types as template arguments. 

Investigate the following.
* Detect whether `DeclRefExpr` points to a class or a constexpr. If it is a constexpr, then find a way to desugar it and get the type?

rseac, 2020-09-08:
Possibly fixed in c1dd9ff51d5c23cb4a09c821b6c16801f47f9933. Can you check?

mayagokhale, 2020-09-09:
Doesn't appear to be changed from before:
object name is c_bplane
HDLtype dump of templatetree args follows
   sc_bv
   fpblk_sz

rseac, 2020-09-09:
Are you using the z3test.cpp?  When I try to run it, I get errors at the beginning...

mayagokhale, 2020-09-09:
It does give error messages.  If you think that is the reason, I will work on fixing that problem first. Don't know why it's complaining about FP, and also I don't know what to do about the __float128 not supported on this target error. But I can work on sc_vector.

rseac, 2020-09-09:
I think that might be a contributing factor. The test I created seems  to work (https://github.com/anikau31/systemc-clang/blob/master/tests/parsing/hotfix-40-findtemplatetypes.cpp#L144) 

@zhuanhao-wu noted that z3test.cpp at dbd753a9fcc44333452cf362de7687cf2880c18c seems to not throw errors. It still has the sc_vector, but no errors.

When using the z3test.cpp from dbd753a9fcc44333452cf362de7687cf2880c18c, I see the following output for c_bplane.

```
   "c_bplane": {
        "array_sizes": [
            64
        ],
        "decl_type": "FieldDecl",
        "is_array_type": true,
        "signal_port_arguments": {
            "sc_bv": [
                "16"
            ],
            "sc_signal": [
                "sc_bv"
            ]
        },
        "signal_port_name": "c_bplane"
    },
```

mayagokhale, 2020-09-09:
I fixed the errors in z3test.cpp, restoring to dbd753a for both z3test.cpp and zhw_encode.h. The only remaining difference is the workaround to remove the break statement in the for loop in encode_ints.
However, it is still not generating the constant for fpblk_sz.
 hSigdecl c_bplane [
        hTypeinfo  NONAME [
          hType array##64 [
            hType sc_signal [
              hType sc_bv [
                hType fpblk_sz NOLIST
              ]
            ]
          ]

I will look at the problem more closely tomorrow.

rseac, 2020-09-09:
I looked at the hdl output I was getting. I see this.

```
     ]
      hSigdecl c_bplane [
        hTypeinfo  NONAME [
          hType array##64 [
            hType sc_signal [
              hType sc_bv [
                hLiteral 16 NOLIST
              ]
            ]
          ]
        ]
      ]
   
```

mayagokhale, 2020-09-09:
Yes!!! I am seeing the same!

closed 2020-09-09 by mayagokhale