anikau31/systemc-clang #157: Template Parm Matcher doesn't find a field

mayagokhale, 2020-09-30:
It appears that the Template Parm Matcher fails to match a field of type bool. The field tlast of type bool in bits_t (file bits.h in shared2) is not being found, and therefore not being emitted in hcode.

When I changed the type of tlast to sc_bv<1>, it gets emitted:

 hTypedef bits_t [
    hTypeTemplateParam D NOLIST
    hTypeField tdata [
      hType sc_bv [
        hType D NOLIST
      ]
    ]
    hTypeField tlast [
      hType sc_bv [
        hLiteral 1 NOLIST
      ]
    ]
  ]

rseac, 2020-10-01:
@mayagokhale  ab2ec00b64bafdfeb353eab13b9abe00d1773d52 provides an example to help update the matcher in `hdl`.

mayagokhale, 2020-10-01:
Thanks, works great!

closed 2020-10-01 by mayagokhale