anikau31/systemc-clang #159: Missing submodule?

mayagokhale, 2020-10-07:
In the encode_block module of zhw_encode.h, there is a submodule declared fwd_lift:
        /*-------- modules --------*/
        fwd_lift<FP> u_xt;
However, that submodule doesn't appear in the submodule list for encode_block:
generate submodule u_encode_block renamed _sc_module_5
submodule count is 0

Perhaps the API has changed?

rseac, 2020-10-07:
Just to clarify, is this when DIM is 1? From what I see in the z3test.cpp, DIM is set to 2, which has fwd_lift<int> as an array. This was something I have been working on. Could you do a merge from master?

mayagokhale, 2020-10-07:
Sorry previously I was looking at the DIM=1 case.

I also just found the example code, so will take a look at that.

rseac, 2020-10-08:
To follow up on this. I tested with DIM=1, and it recognizes the submodule as well. Probably the fix I pushed in recently.

mayagokhale, 2020-10-08:
I must be doing something wrong. I did a pull of master, a merge, re-built and have the same result, that it finds the submodule count for encode_block to be 0, so fwd_lift isn't generated.

rseac, 2020-10-08:
@mayagokhale  I will take a look at this. I think I know why you're experiencing this.

rseac, 2020-12-13:
With the changes in handling arrays for submodules, systemc-clang recognizes the u_xt and y_yt submodule instances now. 
```
{
    "instance_name": "NONE",
    "is_array": "false",
    "module_name": "encode_block",
    "nested_modules": [
        [
            "u_xt_0",
            "u_xt_1",
            "u_xt_2",
            "u_xt_3"
        ],
        [
            "u_yt_0",
            "u_yt_1",
            "u_yt_2",
            "u_yt_3"
        ]
    ],
    "template_args": [
        "struct fp_t<11, 52>",
        "2"
    ]
```

closed 2020-12-14 by rseac