anikau31/systemc-clang #167: port binding question

mayagokhale, 2020-12-18:
In the port bindings, the caller port name is identified as a true port and the caller instance name is shown as the module instance of the calling port. However, it appears that the callee port name is empty, but the callee instance name contains the callee port name:

e.g.
SC port binding found Caller port name  clk caller instance name u_dut <==> callee port name  callee instance name clk

SC port binding found Caller port name  m_block caller instance name u_block_buffer <==> callee port name  callee instance name c_bb_b
lock

Is this the intended representation? Or should the callee port name be populated with the string appearing in the callee instance name?

rseac, 2020-12-20:
In general, the approach I've taken is best illustrated with the following example. Suppose that we have port binding such as: 
```
A.b(C.d);
```
where b and d are ports, and A and C are instances (either modules, ports, etc.)  If everything is working correctly, then:

1. `A` would be the caller instance name.
2. `b` would be the caller port name
3. `C` would be the callee instance name.
4. `d` would be the callee port name

For the examples you mention, assuming that the code for the first example is:
```
u_dut.clk(clk);
```
then, 
1. `u_dut` would be the caller instance name.
2. `clk` would be the caller port name
3. `clk` would be the callee instance name.
4. `` (empty) would be the callee port name

```
u_block_buffer.m_block[i] (c_bb_block[i]));
```
1. `u_block_buffer` would be the caller instance name.
2. `m_block` would be the caller port name
3. `c_bb_block` would be the callee instance name.
4. `` (empty) would be the callee port name

I'm happy to alter and update given any suggestions.

mayagokhale, 2020-12-20:
I don't quite understand the setting of items 3 above. If "instance name" refers to a module instance, neither clk or c_bb_block are module instances, so I would have expected them to be in the callee port name slot.

I also got confused about what is caller and what is callee. In the examples I found on the web, it seems the caller is the submodule declared in the module being processed, and the callee is a prot or signal in the module being processed. So "caller instance name" == "submodule instance name" and "callee instance name" = "thismodule instance name".

So u_dut and u_block_buffer are submodules, "thismodule" isn't given, just the port or signal defined in thismodule scope. Currently the latter is being provided in the "callee instance name" slot. I guess if the "callee instance name" were provided, then the callee port name slot would be populated.

rseac, 2020-12-21:
I agree that the names are not that clear. I'd be happy to update them as it does not affect the matchers much at all.

The instance name doesn't necessarily mean module instance. It could be a port or signal instance as well. So, in the example below:
```
u_dut.clk(clk);
```
The  `clk` would be the callee instance name because it's an instance of a sc_signal.

The terminology for caller and callee is certainly not the best. But, I was having difficulty in finding something meaningful as well. We can change these names.

mayagokhale, 2020-12-21:
No need to change anything. The pattern 1.2(3.4) is mapped to [caller-instance, caller-port, callee-instance, callee-port]. If 4 is not present, callee-port is empty.

closed 2020-12-21 by mayagokhale