anikau31/systemc-clang #169: Arrays as sensitivity list arguments

rseac, 2020-12-21:
Sensitivity list generation with arguments as arrays are not captured.

```
    sensitive << x[stages-1] << y[stages-1] << z[stages-1] << w[stages-1]; 
```

rseac, 2020-12-24:
@mayagokhale Each of the arguments to the `operator<<` call for the sensitivity list is an `ArraySubscriptExpr`. I can certainly provide a pointer to each one of those. Would that be sufficient?

mayagokhale, 2020-12-25:
Pointer to the subscript expression would be fine. I can apply HDLBody to it.

rseac, 2020-12-27:
9e1ae8f includes support for recognizing arrays as arguments for sensitivity lists.

mayagokhale, 2020-12-30:
Since the for loop wasn't being captured, I wasn't able to use that information, similarly to the port bindings in for loops. Now the hcode analysis pass is processing all port bindings and sensitivity lists.

closed 2020-12-30 by mayagokhale