anikau31/systemc-clang #177: SystemC to Verilog Conversion does not work as expected.

myzinsky, 2021-01-01:
When I want to convert the add.cpp as described 

```
#include "systemc.h"
SC_MODULE(topadd2) {
  sc_in_clk clk;
  sc_in<int> in_port_1;
  sc_in<int> in_port_2;
  sc_out<int> out_port;

  SC_CTOR(topadd2) {
    SC_METHOD(topEntry);
    sensitive<<clk.pos();
  }

  void topEntry() {
    out_port.write(in_port_1.read() + in_port_2.read());
  }

};

int sc_main(int argc, char *argv[]){

  sc_clock CLOCK("clock", 5);
  sc_signal<int> input_1;
  sc_signal<int> input_2;
  sc_signal<int> output;
  topadd2 t1 ("t1");

  t1.clk(CLOCK);
  t1.in_port_1(input_1);
  t1.in_port_2(input_2);
  t1.out_port(output);

  sc_start(10, SC_NS);

  return 0;
}
```

The resulting Verilog looks like this:

```verilog
module _sc_module_0 (
  input logic [0:0] clk,
  input logic [31:0] in_port_1,
  input logic [31:0] in_port_2,
  output logic [31:0] out_port
);
endmodule
```

So the entity looks fine, but where is the behaviour?

mayagokhale, 2021-01-01:
Sorry about that. We're in the middle of an enhancement to unroll port bindings and sensitivity lists in "for" loops and the verilog conversion module updates haven't been pushed into master yet. If you back up a few revisions, it should work fine for that example.

We need to do a better job of coordinating these major enhancements.

myzinsky, 2021-01-01:
No Problem! I think this is a really cool project that goes hand in hand with GHDL, YOSYS and many other Open Source tools. It would be great if I could describe HW for my FPGA also with SystemC :-) 

Looking forward!

closed 2022-08-31 by mayagokhale