anikau31/systemc-clang #187: run-compare.py: Difference between `--hdl` and `--golden-intermediate`

rseac, 2021-01-18:
What is the difference between the two flags: `--hdl` and `--golden-intermediate`. Shouldn't we just keep the latter?

zhuanhao-wu, 2021-01-19:
`--hdl` denotes the hdl file when one converts `_hdl.txt` to Verilog.
 `--golden-intermediate` denotes the golden intermediate file when one converts c++ to hdl or c++ to Verilog.

rseac, 2021-01-19:
I find it somewhat confusing. The documentation says:

```
Apart from the conversion, one can specify a golden standard file with --hdl option, and the tool will do a diff at the end of the conversion against the specified golden standard.
```
Maybe a simple way to resolve this would be to explain what each option does in the documentation.

hcy719, 2021-12-09:
I am very interested in this tool, but I always fail to install it. Could you please give me a more detailed installation method

mayagokhale, 2021-12-10:
Did you try the docker image?

hcy719, 2021-12-10:
I have not tried this method. I have tried to install various versions of LLVM, but all failed.Now I try to use Docker Image, but I am not familiar with it, can I directly pull the Docker from this website?rseac/systemc-clang - Docker Image | Docker Hub

zhuanhao-wu, 2022-05-12:
Closed as the script is superceded by scripts from [run-cpp-sv.sh](https://github.com/anikau31/systemc-clang/blob/master/scripts/run-cpp-sv.sh), [run.sh](https://github.com/anikau31/systemc-clang/blob/master/scripts/run.sh), and [run-hcode-sv.sh](https://github.com/anikau31/systemc-clang/blob/master/scripts/run-hcode-sv.sh)

closed 2022-05-12 by zhuanhao-wu