anikau31/systemc-clang #220: "make test" shows many tests fail

mayagokhale, 2021-03-02:
The following tests FAILED:
	  1 - verilog-syntax-error (Failed)
	  2 - verilog-generate-file (Failed)
	  3 - verilog-translation-add (Failed)
	  4 - verilog-translation-z3test (Failed)
	  5 - verilog-translation-moving-average (Failed)
	  7 - instance-matcher (Child aborted)
	  8 - netlist-matcher (Child aborted)
	  9 - netlist-templated-matcher (Child aborted)
	 10 - find-members (Child aborted)
	 11 - sensitivity-matcher (Child aborted)
	 13 - counter (Child aborted)
	 14 - xor-hierarchy (Child aborted)
	 15 - simple-parsing (Child aborted)
	 16 - model-from-file (Child aborted)
	 18 - hotfix-40-findtemplatetypes (Child aborted)
	 19 - template-matching (Child aborted)
	 21 - sreg (Child aborted)
	 22 - portarray-example (Child aborted)
	 23 - inherit (Child aborted)
	 24 - sreg-test (Child aborted)
	 25 - find-emax (Child aborted)
	 26 - test-nested-stream-ports (SEGFAULT)

rseac, 2021-03-03:
All of these tests are running on the CI. So, I'm surprised they are failing.  Can you run `ctest` instead and see if the tests still fail?

mayagokhale, 2021-03-03:
The same happened with ctest (running in the master branch). I wonder what I'm doing wrong.

15% tests passed, 22 tests failed out of 26

Total Test time (real) =  50.50 sec

The following tests FAILED:
	  1 - verilog-syntax-error (Failed)
	  2 - verilog-generate-file (Failed)
	  3 - verilog-translation-add (Failed)
	  4 - verilog-translation-z3test (Failed)
	  5 - verilog-translation-moving-average (Failed)
	  7 - instance-matcher (Child aborted)
	  8 - netlist-matcher (Child aborted)
	  9 - netlist-templated-matcher (Child aborted)
	 10 - find-members (Child aborted)
	 11 - sensitivity-matcher (Child aborted)
	 13 - counter (Child aborted)
	 14 - xor-hierarchy (Child aborted)
	 15 - simple-parsing (Child aborted)
	 16 - model-from-file (Child aborted)
	 18 - hotfix-40-findtemplatetypes (Child aborted)
	 19 - template-matching (Child aborted)
	 21 - sreg (Child aborted)
	 22 - portarray-example (Child aborted)
	 23 - inherit (Child aborted)
	 24 - sreg-test (Child aborted)
	 25 - find-emax (Child aborted)
	 26 - test-nested-stream-ports (SEGFAULT)

rseac, 2021-03-03:
Could you try to run one test? You could follow the steps below.

* Suppose you are in `$SYSTEMC_CLANG_BUILD_DIR`
* You can try to run: `tests/parsing/simple-parsing`

Maybe paste the output ...

mayagokhale, 2021-03-03:
I think it's my problem. It can't find assert.h due to a clang path problem on my mac.

.../systemc-clang-b]$ tests/parsing/simple-parsing 
In file included from input.cc:2:
In file included from /Users/gokhale2/local/systemc-2.3.3//include/systemc.h:43:
/Users/gokhale2/src/llvm/clang+llvm-11.0.0-x86_64-apple-darwin/include/c++/v1/cassert:20:10: fatal error: 'assert.h' file not found
#include <assert.h>
         ^~~~~~~~~~
 ================== INSTANCE MATCHER ================= 
## VD: d 0x7f8b07c01490 instance_name d
=> VD: var_name d var_type_name struct DUT parent_name 
ModuleInstanceMap --  type_decl: 0x7f8b07c002c8 inst_decl: 0x7f8b07c01490 var_type_name: struct DUT var_name: d instance_name: d parent_name:  parent_decl: 0x0 is_field_decl: 0 is_array_: 0
Array sizes: 0 -- 
Array parameters: 0
[x][y][z]: 0,0,0
Inserting VD instance
###### DUMP Instance Matches 
################## INSTANCE MATCHER DUMP 
decl* 0x7f8b07c01490  d instance_field*: 0x7f8b07c002c8
ModuleInstanceMap --  type_decl: 0x7f8b07c002c8 inst_decl: 0x7f8b07c01490 var_type_name: struct DUT var_name: d instance_name: d parent_name:  parent_decl: 0x0 is_field_decl: 0 is_array_: 0
Array sizes: 0 -- 
Array parameters: 0
[x][y][z]: 0,0,0
############### ====> INST: 0x7f8b07c01490, name: DUT, instance_name: d

[findInstance] instance size: 1
find decl name: DUT 0x7f8b07c002c8
- vd 0x7f8b07c002c8  ModuleInstanceMap --  type_decl: 0x7f8b07c002c8 inst_decl: 0x7f8b07c01490 var_type_name: struct DUT var_name: d instance_name: d parent_name:  parent_decl: 0x0 is_field_decl: 0 is_array_: 0
Array sizes: 0 -- 
Array parameters: 0
[x][y][z]: 0,0,0
- Insert vardecl into found instance
=> found_instances: 1
Print out the other fd
 Found field other_fields: sig1
type is : int
(int) int > Template args (DFT): int 
Print out the other fd
 Found field other_fields: double_sig
type is : int
(int) int > Template args (DFT): int 
Print out the other fd
 Found field other_fields: others
type is : int
(int) int > Template args (DFT): int 
========== Port Matcher =============
name: sig1, FieldDecl*: 0x7f8b07c00530(int) int > Template args (DFT): int 

name: double_sig, FieldDecl*: 0x7f8b07c005b0(int) int > Template args (DFT): int 

name: others, FieldDecl*: 0x7f8b07c006d8(int) int > Template args (DFT): int 

Identified submodules
=============================== BASES =======================
== Print module declarations pruned
[DBG] Module instances: 1
CXXRecordDecl* 0x7f8b07c002c8, module name: DUT
################## INSTANCE MATCHER DUMP 
decl* 0x7f8b07c01490  d instance_field*: 0x7f8b07c002c8
ModuleInstanceMap --  type_decl: 0x7f8b07c002c8 inst_decl: 0x7f8b07c01490 var_type_name: struct DUT var_name: d instance_name: d parent_name:  parent_decl: 0x0 is_field_decl: 0 is_array_: 0
Array sizes: 0 -- 
Array parameters: 0
[x][y][z]: 0,0,0
================ END =============== 
############# ============= NEW FIRE ============ ################
Size of module instances: 1
4. Set the constructor.
5. Set the entry functions
6. Set the process
############# =====  END NEW FIRE ============ ################
[... 98 lines cut ...]
-------------------------------------------------------------------------------
/Users/gokhale2/src/systemc-clang-top/systemc-clang/tests/parsing/simple-parsing.cpp:145
...............................................................................

/Users/gokhale2/src/systemc-clang-top/systemc-clang/tests/parsing/simple-parsing.cpp:149: FAILED:
  REQUIRE( instances.size() == 3 )
with expansion:
  1 == 3
with message:
  Checking number of sc_module instances found: 1

Catch will terminate because it needed to throw an exception.
The message was: Test failure requires aborting test!
libc++abi: terminating
/Users/gokhale2/src/systemc-clang-top/systemc-clang/tests/parsing/simple-parsing.cpp:149: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  Checking number of sc_module instances found: 1
  SIGABRT - Abort (abnormal termination) signal

===============================================================================
test cases: 1 | 1 failed
assertions: 2 | 2 failed

Abort trap: 6

mayagokhale, 2021-03-03:
This is due to misconfiguration locally.

closed 2021-03-03 by mayagokhale