anikau31/systemc-clang #292: I am very interested in this tool, but I always fail to install it. Could you please give me a more detailed installation method

hcy719, 2021-12-09:


rseac, 2021-12-16:
Hello,

It would help us if you were to provide more details on how far you got, and what troubles you had.

Is this not helpful: [Installation](https://systemc-clang.readthedocs.io/en/latest/install/install.html)?

hcy719, 2021-12-16:
Thanks for your reply.I followed the installation process and it failed. I guess my biggest problem is how to install the proper LLVM. First of all, I am not sure about the installation method and installation path of LLVM. If I use apt to install, when I cmake, SYSTEMc-clang will not find the AST folder. If I use source compilation, when I do cmake, I won't find the clangconfig. cmake file, but it exists in the path.This problem has bothered me for a long time and I hope to get your help.

mayagokhale, 2021-12-16:
Have you tried exporting the environment variable CLANG_INCLUDE? e.g. if XCode on mac is used:
export CLANG_INCLUDE=$XCODE_PATH/usr/lib/clang/12.0.0/include/

hcy719, 2021-12-18:
> Have you tried exporting the environment variable CLANG_INCLUDE? e.g. if XCode on mac is used: export CLANG_INCLUDE=$XCODE_PATH/usr/lib/clang/12.0.0/include/

I tried, but I still failed. Have you succeeded in MacOS? Is there a more detailed installation process？

ashionjy, 2021-12-18:
> Have you tried exporting the environment variable CLANG_INCLUDE? e.g. if XCode on mac is used: export CLANG_INCLUDE=$XCODE_PATH/usr/lib/clang/12.0.0/include/

I have the same problem as the author. Do you have a MacOS image file? Can you send me an image file and tell me more details about the installation process?

ashionjy, 2021-12-19:
> Please attach a log file showing your build process and the errors generated. That will help us improve the installation guide to handle issues we've not encountered.

Thank you for your reply. The log file is sent to you as an attachment. My installation environment dependencies and tool versions are as follows: cmake is 3.22, gcc is 9.3, g++ is 9.3, llvm is 12.0.1, clang is 12.0.1, and systemc-clang is 2.2.0.

mayagokhale, 2021-12-19:
There wasn't an attachment.

ashionjy, 2021-12-21:
[CMakeOutput.log](https://github.com/anikau31/systemc-clang/files/7749276/CMakeOutput.log)
This is the log file.

mayagokhale, 2021-12-21:
The attachment did get included. However it ends at line 461 and doesn't report an error:
[/usr/lib/gcc/x86_64-linux-gnu/9]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib] ==> [/usr/lib]
  collapse library dir [/lib/x86_64-linux-gnu] ==> [/lib/x86_64-linux-gnu]
  collapse library dir [/lib/../lib] ==> [/lib]
  collapse library dir [/usr/lib/x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/../lib] ==> [/usr/lib]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/9/../../..] ==> [/usr/lib]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit dirs: [/usr/lib/gcc/x86_64-linux-gnu/9;/usr/lib/x86_64-linux-gnu;/usr/lib;/lib/x86_64-linux-gnu;/lib]
  implicit fwks: []

rseac, 2021-12-22:
@ashionjy @hcy719 For the most part, we can use pre-built binaries for LLVM/Clang.  Having said that, it should be possible to build clang/llvm from source for MacOSX.  

Have you tried the pre-built binaries from [here](https://github.com/llvm/llvm-project/releases/tag/llvmorg-13.0.0)?

ashionjy, 2021-12-23:
I tried to use pre-built binary files before, but there was a problem that I couldn't find the header files in the AST. Now I try to use source code to build LLVM/Clang on macOS.

rseac, 2021-12-24:
Can you point me to the binary you used?

ashionjy, 2021-12-28:
The pre-compiled binary I used was 12.0.0, and an error occurred when I executed the command cmake ../systemc-clang. Below are the log files of cmakeerror and cmakeoutput.
[CMakeError.log](https://github.com/anikau31/systemc-clang/files/7783418/CMakeError.log)
[CMakeOutput.log](https://github.com/anikau31/systemc-clang/files/7783419/CMakeOutput.log)

rseac, 2021-12-30:
It appears your CXX compiler is not found.  Could you set the CXX environment variable to g++ or some other c++ compiler and try?

rseac, 2022-02-20:
@ashionjy  I'm assuming it is safe to close this issue?

closed 2022-02-22 by rseac