anikau31/systemc-clang #293: failed with make

dreamqin68, 2021-12-19:
Hi,

I was unable to find the header file when I installed it according to the installation instructions.

```
Scanning dependencies of target utils
[  2%] Building CXX object src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o
In file included from /home/user/systemc-clang/src/utils/ArrayTypeUtils.cpp:1:
/home/dream/systemc-clang/src/utils/ArrayTypeUtils.h:5:10: fatal error: 'clang/AST/Type.h' file not found
#include "clang/AST/Type.h"
         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/utils/CMakeFiles/utils.dir/build.make:63: src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1105: src/utils/CMakeFiles/utils.dir/all] Error 2
make: *** [Makefile:95: all] Error 2
```
I set the llvm installation path in `script/paths.sh`.
OS: ubuntu 20.04.3 LTS
clang: 12.0.0

rseac, 2021-12-22:
Hello @dreamqin68,

This is because clang include headers are not being found. 

1. Could you tell us how you installed clang/llvm?  Did you use the pre-built binaries?
2. Could you try to explicitly put the location for the LLVM/clang using the LLVM_INSTALL_DIR flag when running cmake to build systemc-clang?

dreamqin68, 2021-12-22:
Hi @rseac,

1. I built `LLVM/Clang` from the source according to the [installation instructions](https://systemc-clang.readthedocs.io/en/latest/install/install.html). 
2. I put `export LLVM_INSTALL_DIR=/home/user/llvm-project/build/` in `script/paths.sh`

rseac, 2021-12-22:
The issue seems to be that the include path to find $LLVM_INSTALL_DIR/include/ is not being set correctly.

1. After sourcing scripts/path.sh could you echo out the following variables and paste them to me.

* LVM_CXX_FLAGS 
* LLVM_LIBS 
* LLVM_LD_FLAGS 

2. Can you enable verbose output for make?  make VERBOSE=1.  Please attach the output.

dreamqin68, 2021-12-23:
1. 
```
echo $LLVM_CXX_FLAGS
-I/home/user/llvm-project/llvm/include -I/home/user/llvm-project/build/include -std=c++14 -fno-exceptions -fno-rtti -D_GNU_SOURCE -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -fno-aligned-allocation -fvisibility-inlines-hidden
```
2.
```
echo $LLVM_LIBS
-lLLVMWindowsManifest -lLLVMXRay -lLLVMLibDriver -lLLVMDlltoolDriver -lLLVMCoverage -lLLVMLineEditor -lLLVMXCoreDisassembler -lLLVMXCoreCodeGen -lLLVMXCoreDesc -lLLVMXCoreInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMX86Desc -lLLVMX86Info -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyInfo -lLLVMSystemZDisassembler -lLLVMSystemZAsmParser -lLLVMSystemZCodeGen -lLLVMSystemZDesc -lLLVMSystemZInfo -lLLVMSparcDisassembler -lLLVMSparcAsmParser -lLLVMSparcCodeGen -lLLVMSparcDesc -lLLVMSparcInfo -lLLVMRISCVDisassembler -lLLVMRISCVAsmParser -lLLVMRISCVCodeGen -lLLVMRISCVDesc -lLLVMRISCVInfo -lLLVMPowerPCDisassembler -lLLVMPowerPCAsmParser -lLLVMPowerPCCodeGen -lLLVMPowerPCDesc -lLLVMPowerPCInfo -lLLVMNVPTXCodeGen -lLLVMNVPTXDesc -lLLVMNVPTXInfo -lLLVMMSP430Disassembler -lLLVMMSP430AsmParser -lLLVMMSP430CodeGen -lLLVMMSP430Desc -lLLVMMSP430Info -lLLVMMipsDisassembler -lLLVMMipsAsmParser -lLLVMMipsCodeGen -lLLVMMipsDesc -lLLVMMipsInfo -lLLVMLanaiDisassembler -lLLVMLanaiCodeGen -lLLVMLanaiAsmParser -lLLVMLanaiDesc -lLLVMLanaiInfo -lLLVMHexagonDisassembler -lLLVMHexagonCodeGen -lLLVMHexagonAsmParser -lLLVMHexagonDesc -lLLVMHexagonInfo -lLLVMBPFDisassembler -lLLVMBPFAsmParser -lLLVMBPFCodeGen -lLLVMBPFDesc -lLLVMBPFInfo -lLLVMAVRDisassembler -lLLVMAVRAsmParser -lLLVMAVRCodeGen -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMARMDisassembler -lLLVMARMAsmParser -lLLVMARMCodeGen -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMAMDGPUDisassembler -lLLVMAMDGPUAsmParser -lLLVMAMDGPUCodeGen -lLLVMAMDGPUDesc -lLLVMAMDGPUUtils -lLLVMAMDGPUInfo -lLLVMAArch64Disassembler -lLLVMAArch64AsmParser -lLLVMAArch64CodeGen -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMOrcJIT -lLLVMMCJIT -lLLVMJITLink -lLLVMOrcTargetProcess -lLLVMOrcShared -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMSymbolize -lLLVMDebugInfoPDB -lLLVMDebugInfoGSYM -lLLVMOption -lLLVMObjectYAML -lLLVMMCA -lLLVMMCDisassembler -lLLVMLTO -lLLVMPasses -lLLVMCFGuard -lLLVMCoroutines -lLLVMObjCARCOpts -lLLVMHelloNew -lLLVMipo -lLLVMVectorize -lLLVMLinker -lLLVMInstrumentation -lLLVMFrontendOpenMP -lLLVMFrontendOpenACC -lLLVMExtensions -lLLVMDWARFLinker -lLLVMGlobalISel -lLLVMMIRParser -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMSelectionDAG -lLLVMCodeGen -lLLVMIRReader -lLLVMAsmParser -lLLVMInterfaceStub -lLLVMFileCheck -lLLVMFuzzMutate -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMBinaryFormat -lLLVMTableGen -lLLVMSupport -lLLVMDemangle

```
3.
```
echo $LLVM_LD_FLAGS
-L/home/user/llvm-project/build/lib
```
4. 
```
make VERBOSE=1
/usr/bin/cmake -S/home/user/systemc-clang -B/home/user/systemc-clang-build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/user/systemc-clang-build/CMakeFiles /home/user/systemc-clang-build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/user/systemc-clang-build'
make -f src/utils/CMakeFiles/utils.dir/build.make src/utils/CMakeFiles/utils.dir/depend
make[2]: Entering directory '/home/user/systemc-clang-build'
cd /home/user/systemc-clang-build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/systemc-clang /home/user/systemc-clang/src/utils /home/user/systemc-clang-build /home/user/systemc-clang-build/src/utils /home/user/systemc-clang-build/src/utils/CMakeFiles/utils.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/user/systemc-clang-build'
make -f src/utils/CMakeFiles/utils.dir/build.make src/utils/CMakeFiles/utils.dir/build
make[2]: Entering directory '/home/user/systemc-clang-build'
[  2%] Building CXX object src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o
cd /home/user/systemc-clang-build/src/utils && /usr/bin/clang++  -DJSON_EXCEPTION -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -I/home/user/llvm-project/build/include -I/home/user/systemc-clang/externals -I/home/user/systemc-clang/src -I/home/user/systemc-clang/src/matchers -I/home/user/systemc-clang/src/model -I/home/user/systemc-clang/src/cfg -I/home/user/systemc-clang/src/utils -I/home/user/systemc-clang/src/SCuitable  -isystem/home/user/llvm-project/llvm/include -isystem/home/user/llvm-project/build/include -std=c++14   -fno-exceptions -fno-rtti -D_GNU_SOURCE -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS  -fvisibility=hidden -fvisibility-inlines-hidden  -iquote /home/user/systemc-clang/src -O3   -std=c++17 -o CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o -c /home/user/systemc-clang/src/utils/ArrayTypeUtils.cpp
In file included from /home/user/systemc-clang/src/utils/ArrayTypeUtils.cpp:1:
/home/user/systemc-clang/src/utils/ArrayTypeUtils.h:5:10: fatal error: 'clang/AST/Type.h' file not found
#include "clang/AST/Type.h"
         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/utils/CMakeFiles/utils.dir/build.make:63: src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o] Error 1
make[2]: Leaving directory '/home/user/systemc-clang-build'
make[1]: *** [CMakeFiles/Makefile2:1105: src/utils/CMakeFiles/utils.dir/all] Error 2
make[1]: Leaving directory '/home/user/systemc-clang-build'
make: *** [Makefile:95: all] Error 2
```

rseac, 2021-12-24:
I believe the issue is that the LLVM_INSTALL_DIR might need to be set differently.  

Could you try setting LLVM_INSTALL_DIR to:

LLVM_INSTALL_DIR=/home/user/llvm-project/

dreamqin68, 2021-12-24:
After `export LLVM_INSTALL_DIR=/home/user/llvm-project/`
```
source scripts/paths.sh 
bash: /home/user/llvm-project//bin/llvm-config: No such file or directory
```

rseac, 2021-12-24:
I noticed that the build targets for LLVM/clang has changed from source. 

Can you help me by first trying out the build from binaries before we determine how to get it to work with with the source?

1. Could you install the corresponding binary for your platform?
2. Build systemc-clang as suggested?

Let me know which binary you used.

dreamqin68, 2021-12-24:
I installed `llvm-12` for `ubuntu 20.04.3 LTS`.
I put `export LLVM_INSTALL_DIR=/usr/lib/llvm-12/` in `script/paths.sh`.
The `llvm-config` binary is `/usr/lib/llvm-12/bin/llvm-config`.

After `make`, there is still the same error here.
```
Scanning dependencies of target utils
[  2%] Building CXX object src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o
In file included from /home/user/systemc-clang/src/utils/ArrayTypeUtils.cpp:1:
/home/user/systemc-clang/src/utils/ArrayTypeUtils.h:5:10: fatal error: 'clang/AST/Type.h' file not found
#include "clang/AST/Type.h"
         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/utils/CMakeFiles/utils.dir/build.make:63: src/utils/CMakeFiles/utils.dir/ArrayTypeUtils.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1105: src/utils/CMakeFiles/utils.dir/all] Error 2
make: *** [Makefile:95: all] Error 2
```

rseac, 2021-12-25:
Could I request you to try the pre-built binaries and published by llvm/clang instead of the package manager from the Ubuntu distribution?  

The below is the pre-built binary I use on the same host as you mentioned.

[https://github.com/llvm/llvm-project/releases/tag/llvmorg-12.0.0](https://github.com/llvm/llvm-project/releases/tag/llvmorg-12.0.0)

dreamqin68, 2021-12-26:
Systemc-clang was successfully installed by the pre-built `llvm/clang` binaries, thanks for your help.

rseac, 2021-12-27:
Great to hear that.

closed 2021-12-27 by rseac