anikau31/systemc-clang #326: Reduction functions from SystemC should have their own hNode

zhuanhao-wu, 2022-05-09:
# Description
In [test_sc_types.cpp](https://github.com/intel/systemc-compiler/blob/main/tests/method/test_sc_types.cpp#L291), there are reduction functions `and_reduce`, `or_reduce`, `xor_reduce`, `nand_reduce`, `nor_reduce`, and `xnor_reduce`.

Currently, they are generated as functions in the hCode file without bit width information:
```
hMethodCall sc_dt__or_reduce_func_1 [
    hVarref x_sc_uint_reduce_local_56 NOLIST
]
...
hFunction sc_dt__or_reduce_func_1 [
  hFunctionRetType  NONAME [
    hTypeinfo  NONAME [
      hType _Bool NOLIST
    ]
  ]
  hFunctionParams  NONAME [
    hFunctionParamIO a [
      hTypeinfo  NONAME [
        hType sc_uint_base NOLIST
      ]
    ]
  ]
  hCStmt  NONAME [
    hReturnStmt  NONAME [
      hNoop or_reduce [
        hVarref a NOLIST
      ]
    ]
  ]
]
```

We could have special hUnop for these functions, for example, with the following form:

```
hUnop or_reduce [
    hVarref x_sc_uint_reduce_local_56 NOLIST
]
```

mayagokhale, 2022-05-24:
This issue has been addressed by the latest commit to scratchllnl. A new hcode hBuiltinFunction was added with the builtin function name as argument. In addition to the reduce functions, range and bit are also recognized. The commit breaks CI due to the new opcode.

This issue can be closed once the generated hcode has been verified and the verilog parser updated.

zhuanhao-wu, 2022-05-24:
The verilog generator is updated with #334

closed 2022-05-24 by zhuanhao-wu