anikau31/systemc-clang #327: Thread state machine has malformatted if statement

zhuanhao-wu, 2022-05-09:
# Description

The state machine for `threadProc()` thread in [test_const_prop_loop.cpp](https://github.com/intel/systemc-compiler/blob/main/tests/method/test_const_prop_loop.cpp#L67) contains `hIfStmt` node that has `hCStmt NONAME NOLIST` as its condition.

The `threadProc()` thread:
```c++
void threadProc() 
{
    unsigned slot = 0;
    wait();
    
    while (true) 
    {
        slot = (slot == 2) ? 0 : slot + 1;
        wait();
    } 
}
```

The generated switch case branch:
```
hSwitchCase  NONAME [
  hLiteral 1 NOLIST
  hIfStmt  NONAME [
    hLiteral 1 NOLIST
    hCStmt  NONAME NOLIST
  ]	
  hIfStmt  NONAME [
    hCStmt  NONAME NOLIST
  ]
  hLiteral 0 NOLIST
  hBinop = [
    hVarref slot_threadProc_thread_0 NOLIST
    hCondop  NONAME [
      hBinop == [
        hVarref slot_threadProc_thread_0 NOLIST
        hLiteral 2 NOLIST
      ]
      hLiteral 0 NOLIST
      hBinop + [
        hVarref slot_threadProc_thread_0 NOLIST
        hLiteral 1 NOLIST
      ]
    ]
  ]
  hBinop = [
    hVarref _next_state_threadProc# NOLIST
    hLiteral 2 NOLIST
  ]
  hReturnStmt  NONAME NOLIST
  hBinop + [
    hVarref slot_threadProc_thread_0 NOLIST
    hLiteral 1 NOLIST
  ]
]
```

mayagokhale, 2022-05-11:
SplitCFG includes subexpressions of conditional expression as split graph blocks, and plugin is generating hcode for these blocks.

zhuanhao-wu, 2022-05-16:
[image]