anikau31/systemc-clang #365: systemc-clang make project fails in src/matchers/FindTemplateTypes.cpp

KarimAlatrash, 2022-09-19:
# Description
When following installation instructions [here](https://systemc-clang.readthedocs.io/en/latest/install/install.html#installing-systemc-clang-on-osx) from scratch, running `make` in the system-clang-build folder throws the following error:
```
/Users/karim/Developer/SystemCVerilog/systemc-clang/src/matchers/FindTemplateTypes.cpp:103:40: error: cannot initialize a parameter of type 'clang::TemplateName::Qualified' with an rvalue of type 'int'
  template_name.print(sstream, Policy, 0);
                                       ^
/Users/karim/Developer/SystemCVerilog/llvm-project/build/../clang/include/clang/AST/TemplateName.h:335:24: note: passing argument to parameter 'Qual' here
             Qualified Qual = Qualified::AsWritten) const;
                       ^
1 error generated.
make[2]: *** [src/matchers/CMakeFiles/matchers.dir/FindTemplateTypes.cpp.o] Error 1
make[1]: *** [src/matchers/CMakeFiles/matchers.dir/all] Error 2
make: *** [all] Error 2
```

# Reproducing the bug 
* Install latest llvm-project from github following instructions here: [llvm install instructions](https://clang.llvm.org/get_started.html)
* Install latest systemC from github using CMake instructions found here: [systemc install instructions](https://github.com/accellera-official/systemc/blob/master/cmake/INSTALL_USING_CMAKE)
* Follow installation steps found here: [systemc-clang install instructions](https://github.com/accellera-official/systemc/blob/master/cmake/INSTALL_USING_CMAKE)
* Upon running make, you should observe the error above.

## Versions 
- systemc-clang: Latest commit on master found [here](https://github.com/anikau31/systemc-clang/commit/d3641638d81e4dc1fedc839cc2f56b8f99589877)
- CXX compiler: clang-16.0.0git
- CXX flags: N/A
- SystemC: Latest commit on master found [here](https://github.com/accellera-official/systemc/commit/4a44d3619eca48a7a99122ae8c0fe3baf101d2e5) which is at least 2.3.4
- Operating System: macOS Monterey, M1 Pro chip (ARM64 architecture)

### Expected behaviour
The project `make` command should compile the project without errors.

### Actual behaviour
The project `make` command stops at the error shown above

KarimAlatrash, 2022-09-19:
Currently investigating a workaround involving switching from clang-16.0.0 to clang13.0.0 and rebuilding the project

KarimAlatrash, 2022-09-19:
Update: Using clang 13.0.0 fixes the issue

rseac, 2023-08-28:
Fixed in 9093b0f57d327b6cd1953e1decb81badb1d15b4

closed 2023-08-28 by rseac