anikau31/systemc-clang #368: test_level2.cpp fails

zhuanhao-wu, 2022-10-11:
The systemc-clang binary crashes, it does not seem to be a syntax error.
The design is a mix of thread and processes.

https://github.com/intel/systemc-compiler/blob/main/tests/method/test_level2.cpp

The last couple lines of log:

```
systemc-clang: /opt/clang-13.0.0/include/llvm/ADT/SmallVector.h:281: const T& llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::operator[](llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type) const [with T = std::pair<const systemc_clang::SplitCFGBlock*, systemc_clang::SupplementaryInfo>; <template-parameter-1-2> = void; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::const_reference = const std::pair<const systemc_clang::SplitCFGBlock*, systemc_clang::SupplementaryInfo>&; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type = long unsigned int]: Assertion `idx < size()' failed.
Aborted (core dumped)
```

rseac, 2022-11-08:
Started digging into this.  It appears to be within the `break_in_while_for` thread process.

rseac, 2022-11-11:
It appears that 4638531fbc9295a61a1b02c2b2d861e11fd5a9de fixes the crash.  Essentially, correctly identifying IF blocks in the SplitCFG.

rseac, 2022-11-13:
This should be fixed in scratchllnl.

rseac, 2023-02-05:
Confirmed that it does not crash on current scratchllnl. Closing.

commit dc328568 (2022-11-13): Fix: Correct the index for the false path for if blocks.

The issue was somewhat specific as follows.

When there is an IF block, and the successor for the false is a block
that has been visited, the traversal reaches the IF block at the end,
and notices that there is no unvisited successor.  Consequently,
the false path index saved is the size of the currently visited
blocks, which is incorrect.  Instead, we need to find the false path
block in the currently visited blocks.

This should fix #368  and #373
closed 2023-02-05 by rseac