anikau31/systemc-clang #373: Frontend crashes on an example

zhuanhao-wu, 2022-10-31:
https://github.com/intel/systemc-compiler/blob/main/tests/cthread/test_break.cpp

The last few lines of the errors are:
```
systemc-clang: /opt/clang-13.0.0/include/llvm/ADT/SmallVector.h:281: const T& llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::operator[](llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type) const [with T = std::pair<const systemc_clang::SplitCFGBlock*, systemc_clang::SupplementaryInfo>; <template-parameter-1-2> = void; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::const_reference = const std::pair<const systemc_clang::SplitCFGBlock*, systemc_clang::SupplementaryInfo>&; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type = long unsigned int]: Assertion `idx < size()' failed.
```

This happens with `break_in_for_wait1` (removed sct_asserts):
```
    void break_in_for_wait1()
    {
        int k = 0;
        wait();

        while (1) {

            for (int i = 0; i < 3 ; i++) {  // B6
                k = 2;  // B5
                wait();                 // 1

                if (in.read() > 1) {

                    break;      // B4
                }
            }       // B3
            k = 1;  // B2
            wait();                     // 2
        }
    }
```

rseac, 2022-11-13:
This should be fixed in scratchllnl.

rseac, 2023-02-05:
Confirmed this does not crash on scratchllnl.  Closing.

closed 2023-02-05 by rseac