anikau31/systemc-clang #380: Seg fault on if conditions

zhuanhao-wu, 2023-01-30:
https://github.com/intel/systemc-compiler/blob/main/designs/tests/cthread/test_if.cpp

The following code causes segmentation faults in binary
```c++
static const bool   CONST_A = 1;
...
void smem_if_binary_const() {
    bool b = 0;
    wait();

    while (true) {          // B5
        if (CONST_A) {      // B4
            b = !b;         // B3
        }

        wait();             // B2
    }                       // B1
}
```

Last lines of binary output
```
==============================================
Recurse DFS starting at BB 3 parent BB# 4 T 1 F 0
captured      blocks 5  4
loop visited  blocks 5  4
visited       blocks 5  4  3
local_path   5 4
Visit Parent BB# 3
Found successor BB 2 for parentBB 3
Visit Parent BB# 2
 has WAIT
@@@@@ AFTER DFS for BB# 3 for parent BB# 4 T 1 F 0
captured      blocks 5  4
loop visited  blocks 5  4  3  2  1
visited       blocks 5  4  3
sub_path     3 2
END Recurse DFS at BB 3 parent BB# 4 T 1 F 0
TRUE PATH DIFF BB# 4 curr_path size is 4 5 4 3 2 Segmentation fault (core dumped)
```

rseac, 2023-01-30:
I can't seem to reproduce this.

I created the following test based on the issue: https://github.com/anikau31/systemc-clang/blob/scratchllnl/tests/data/thread-if-380-input.cpp.

What should I change to reproduce the segfault?

rseac, 2023-02-01:
The fix for this was in ensuring that a successor pointer is not null before proceeding to access its members.

closed 2023-02-01 by rseac