anikau31/systemc-clang #386: `wait(int)` support

zhuanhao-wu, 2023-02-13:
Currently, we don't support `wait(n)`: https://github.com/intel/systemc-compiler/blob/main/designs/tests/cthread/test_cthread_wait_n.cpp#L199

This could result in the binary exit abnormally.

```
    void thread2 () {
        while (1) {
            wait(2*2);
            wait(1 + 1 + 1);
        }
    }
```

rseac, 2023-02-13:
I confirm that this does indeed result in an segfault. The culprit is in the AST node parsing of the argument. 

```
Number of entries in split_elements 2
Element number 0 *************** FIRST ARG ****************
BinaryOperator 0x4dfaaa8 'int' '*'
|-IntegerLiteral 0x4dfaa68 'int' 2
`-IntegerLiteral 0x4dfaa88 'int' 2
 ARG VAL:
Program received signal SIGSEGV, Segmentation fault.
clang::APNumericStorage::getIntValue (this=0x10) at /home/twiga/bin/clang-13.0.0/include/clang/AST/Expr.h:1455
1455        unsigned NumWords = llvm::APInt::getNumWords(BitWidth);
(gdb) bt
#0  clang::APNumericStorage::getIntValue (this=0x10) at /home/twiga/bin/clang-13.0.0/include/clang/AST/Expr.h:1455
#1  0x00000000006e0a23 in clang::APIntStorage::getValue (this=0x10) at /home/twiga/bin/clang-13.0.0/include/clang/AST/Expr.h:1466
#2  0x0000000002176a76 in systemc_clang::SplitCFG::getWaitArgument (this=0x7fffffff9b20, element=...) at /home/twiga/code/github/rseac/systemc-clang/src/cfg/SplitCFG.cpp:587
```

rseac, 2023-02-13:
This should be fixed now 4c9eef5.  I made sure to constant fold the expression, if it can be done.  So, for a subset of expressions that are constant foldable, this should be fine.  We still assume the wait's argument evaluates to an integer.

closed 2023-02-13 by rseac