anikau31/systemc-clang #388: Support for `wait()` in (non-thread) function

zhuanhao-wu, 2023-02-13:
https://github.com/intel/systemc-compiler/blob/main/designs/tests/cthread/test_cthread_loop_fail7.cpp

The following thread causes the binary to exit abnormally:
```c++
class top : sc_module
{
public:
    sc_in_clk clk;
    sc_signal<bool> arstn;

    SC_HAS_PROCESS(top);
    top(sc_module_name)
    {
        SC_CTHREAD(test_thread, clk.pos());
        async_reset_signal_is(arstn, false);
    }

    sc_vector<sc_signal<int>> s{"s", 4};

    // Example from DME
    int getNextLdSlot(int curSlot)
    {
        while (true) {
            int nextSlot = curSlot;
            for (unsigned i = 0; i < 4; ++i) {
                if (!s[nextSlot])
                    return nextSlot;

                nextSlot = (nextSlot == 4 - 1) ? 0 : nextSlot + 1;
            }
            wait();
        }
    }

    void test_thread()
    {
        int slot = 0;
        wait();

        while (1)
        {
            slot = getNextLdSlot(slot);
            wait();
        }
    }
};
```

The error output:
```
isNamespace
@@ name is sc_core for @@@@ isSCT
isRecordType
isNamespace
@@ name is sc_core for @@@@ isSCT
isRecordType
isNamespace
@@ name is sc_core for ### CHECK1: old 1 != new 0
systemc-clang: /systemc-clang/plugins/hdl/HDLType.cpp:143: bool HDLType::checkusertype(systemc_clang::TreeNode<systemc_clang::TemplateType>* const&, systemc_clang::Tree<systemc_clang::TemplateType>* const&, std::string&): Assertion `0' failed.
```

zhuanhao-wu, 2023-02-13:
- check if this type of `wait()` is supported in the standard

rseac, 2023-02-20:
At the moment, the SplitCFG construction does not flatten the CFG or inline the calls within the CFGBlock.  I have to add this support to make the SCCFG construction correct.

zhuanhao-wu, 2023-02-20:
Standard: https://www.accellera.org/images/downloads/standards/systemc/SystemC_Synthesis_Subset_1_4_7.pdf

My interpretation of the standard is that this kind of `wait()` is not supported, according to Section 4.2.2 (p. 20)

> The thread process body of a design shall also follow
this structure. Among the available constructs to suspend a process, a design shall only use a
call to wait(), where the wait condition is the clock edge to which the thread process is
sensitive, as specified in the module constructor. That is, the only Supported waits are the
following.
> - wait()
> - wait(int): where the integer argument is statically determinable.
> 
> Other forms of suspending a thread process are Not Supported.

I tend to categorize the `wait()` in another function as _Other forms_.
> ...
>

zhuanhao-wu, 2023-02-20:
Another similar design: [/systemc-compiler/tests/cthread/test_fcall_loop_fail.cpp](https://github.com/intel/systemc-compiler/blob/main/designs/tests/cthread/test_fcall_loop_fail.cpp)

rseac, 2023-02-21:
Identify what is needed to implement the flattening of the CFG, and assign priority.

rseac, 2023-07-18:
Not going to support as yet.  If there is interest from other parties, then we can work on it.

closed 2023-07-18 by rseac