anikau31/systemc-clang #389: Null function body not handled

mayagokhale, 2023-02-13:
https://github.com/intel/systemc-compiler/blob/main/designs/tests/method/test_fcall.cpp

globFunc1 has no function body, thus in the function definition hcode, the body is omitted. However, an empty body is not accepted by hcode parser. Should the hcode plugin generate a fake return stmt or should the hcode parser accept a null body?

zhuanhao-wu, 2023-02-13:
I think it suffices to put an empty hcstmt node, specifically something like this:
```
hCStmt  NONAME NOLIST
```

mayagokhale, 2023-02-14:
fixed in 5c2c1fe

zhuanhao-wu, 2023-02-26:
Currently, a hVardecl is generated within the function body, which should be outside:
```
      hFunction globFunc1_func_10 [
        hFunctionRetType  NONAME [
          hTypeinfo  NONAME [
            hType void NOLIST
          ]
        ]
        hCStmt  NONAME [
          hVardecl a__local_32 [
            hTypeinfo  NONAME [
              hType int NOLIST
            ]
          ]
          hCStmt  NONAME [
            hVarAssign  NONAME [
              hVarref a__local_32 NOLIST
              hLiteral 0 NOLIST
            ]
          ]
        ]
      ]
```
It should be
```
      hFunction globFunc1_func_10 [
        hFunctionRetType  NONAME [
          hTypeinfo  NONAME [
            hType void NOLIST
          ]
        ]
          hVardecl a__local_32 [
            hTypeinfo  NONAME [
              hType int NOLIST
            ]
          ]
        hCStmt  NONAME [

          hCStmt  NONAME [
            hVarAssign  NONAME [
              hVarref a__local_32 NOLIST
              hLiteral 0 NOLIST
            ]
          ]
        ]
      ]
```