anikau31/systemc-clang #390: Long `?:` expression causes binary to execute for an extended time, and causes the generation of excessive `read`.

zhuanhao-wu, 2023-02-21:
This design is in a thread example: https://github.com/intel/systemc-compiler/blob/main/designs/tests/cthread/test_cthread_array.cpp#L91

The following is extracted from the original design, which takes around 26 seconds to run. When the long `?:` is placed within a method, it only takes 3 seconds to run.

```
#include <systemc.h>

// Array general cases and some special cases
class top : sc_module
{
public:
    sc_in<bool>     clk;
    sc_signal<bool> arstn;

    sc_signal<bool> a{"a"};
    sc_signal<bool> ps_repl;

    static const unsigned DOMAIN_NUM = 1;

    SC_HAS_PROCESS(top);
    top(sc_module_name)
    {
        SC_CTHREAD(bug_in_array_index2, clk.pos());
        async_reset_signal_is(arstn, false);
    }

    // BUG in real design -- in ConfigCtrl.h extra copy of @reqSource in Verilog
    // Shorten version
    sc_signal<sc_uint<2>>   fifo_source;
    sc_signal<sc_uint<2>>   master_resp_data[3];
    static const unsigned ECC_NUM           = 0;

    // BUG in real design -- in ConfigCtrl.h extra copy of @reqSource in Verilog
    // Full version
    void bug_in_array_index2() {
	wait();

	while(1) {
            sc_uint<2> reqSource = fifo_source.read();
            bool reqOper = a.read();
            bool reqReg;

            master_resp_data[reqSource] = (
                    (ECC_NUM > 0) ? (sc_uint<1>)a :
                                    (sc_uint<1>)0,
                    (ECC_NUM > 0) ? (sc_uint<1>)a :
                                    (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                    (sc_uint<1>)0,
                    (sc_uint<5>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                    (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                     (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                    (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                   (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                       (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                       (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                     (sc_uint<1>)0,
                    (ps_repl) ? (sc_uint<1>)ps_repl.read() :
                                       (sc_uint<1>)0,
                    (sc_uint<1>)reqOper,
                    (sc_uint<1>)reqOper);
        }

    }
};

int sc_main(int argc, char *argv[])
{
    top top_inst{"top_inst"};
    sc_start(100, SC_NS);
    return 0;
}
```

Currently, excessive orphan reads are generated for `ps_repl` in the hCode, while they do not affect the functionality,  it would be better if they were not present.

```
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hSigAssignR read [
                    hVarref ps_repl_scclang_global_4 NOLIST
                  ]
                  hBinop = [
                    hBinop ARRAYSUBSCRIPT [
                      hVarref master_resp_data_scclang_global_3 NOLIST
                      hVarref reqSource NOLIST
                    ]
                    hBinop concat [
                      hBinop concat [
                        hBinop concat [
                          hBinop concat [
```

zhuanhao-wu, 2023-02-21:
- Try without HDL Plugin

rseac, 2023-06-05:
I can confirm that the issue has to do with the traversal algorithm that generates the path information.

rseac, 2023-07-03:
If we are to assume that there are no `wait()` statements within ternary operators, then we could simply acknowledge in the traversal that there won't be a `wait()` in it; hence, there is no need to do a recursive DFS.  This would provide the same paths without having to repeatedly call the DFS.

rseac, 2023-07-12:
Fixed in #3657b94

closed 2023-07-12 by rseac