anikau31/systemc-clang #401: target test_z3_behav fail on fccm-ae beanch

liu-xueyang, 2023-06-17:
# Description
I'm trying to run behavioral simulation for ZHW on fccm-ae branch using target test_z3_behav.
```
make test_z3_behav
ERROR: [Vivado 12-172] File or Directory '/home3/shared/liu91/systemc-clang/hardware/build/verif/../rtl/sysc_z3test.sv' does not exist​
'/home3/shared/liu91/systemc-clang/hardware/build/verif/../rtl/sysc_z3test.sv' doesn't seems to generate.
```

1. To resolve this I tried to edit sysc_files variable in build/: `verif/test_z3_behav/run_behavioral.test_z3.tcl` and `verif/test_z3_behav/files.tcl` to point to `$REPO_ROOT/hardware/rtl/z3test_hdl.txt.sv​`

2. Then I ran into `invalid command name "::tclapp::xilinx::projutils::ext_export_simulation"​`. I replaced the command with export_simulation​ and added `-extra_xvlog_opt "-L uvm --define ultra96v2_20221" -extra_xelab_opt "-L uvm" ` options manually to `tb.sh`.

3. Finally I'm getting a hang with the simulation step in `tb.sh`
```
WARNING: 5 ns tb.dut.sw.bd_i.zynq_ultra_ps_e_0.inst.S_AXI_HPC1_FPD.slave.IF : ARESET_N can't be X/Z after 1 cycle of clock. To downgrade, use <hierarchy_path to VIP>.IF.set_enable_xchecks_to_warn(), or filter using clr_enable_xchecks.
[165] : *ZYNQ_MPSoC_BFM_INFO : POR and STRB Reset called for 0x0
[165] : *ZYNQ_MPSoC_BFM_INFO : FPGA Soft Reset called for 0xf
[845] : *ZYNQ_MPSoC_BFM_INFO : POR and STRB Reset called for 0x1
[845] : *ZYNQ_MPSoC_BFM_INFO : FPGA Soft Reset called for 0x0
[845] : *ZYNQ_MPSoC_BFM_INFO : Starting Address(0x0) -> DDR Memory is pre-loaded with 524288 bytes of data
                 845 DDR data width is 32 start addr is 4000 
 pre_load_mem_from_file11 entered
[845] : *ZYNQ_MPSoC_BFM_INFO : Starting Address(0x10000) -> DDR Memory is pre-loaded with 8192 bytes of data from file ../../../data/test_zynq_blank.mem
[2125] : M_AXI_HPM0_FPD : *ZYNQ_MPSoC_BFM_INFO : Starting Address(0xa0000000) -> AXI Write -> 4 bytes
^CINFO: [Common 17-41] Interrupt caught. Command should exit soon.
```

liu-xueyang, 2023-06-27:
Update on the RTL bug: using non-blocking (<=) assignments for m_bits_valid therefore signal stays zero the entire test. I believe only one of the following statements are picked up by vivado.
https://github.com/anikau31/systemc-clang/blob/fccm-ae/hardware/rtl/z3test_hdl.txt.sv#L1554
https://github.com/anikau31/systemc-clang/blob/fccm-ae/hardware/rtl/z3test_hdl.txt.sv#L1732
I replaced with blocking assignments (=) for both and waveform looks correct. Similar issues with m_bits_data_tdata and m_bits_data_tlast not resetting properly.

zhuanhao-wu, 2023-07-11:
Thanks for reporting on these issues!

Regarding the tcl script/behavioural simulation errors, they might be a result of the behavioural simulation scripts are not updated properly for the AE branch. I will try to update them.

I can confirm that the blocking vs non-blocking assignments are causing incorrect behavioural simulation results. Currently I suspect that this is related to the behavioural differences between xsim versions. I will try to look into it in more details and figure out a fix. Challenge here would be to ensure consistent behaviour between synthesis and behavioural.