anikau31/systemc-clang #402: How to generate IR for systemc by using this tool?

Elwin-zhao, 2023-08-24:
# Description
A clear and concise description of what the bug is.

# Reproducing the bug 

## Versions 
- systemc-clang: [i.e. 0.1]
- CXX compiler: [i.e. clang-7.0.0]
- CXX flags: [i.e. llvm-config --cxxflags]
- SystemC: [i.e. 2.3.3]
- Operating System: [i.e. Ubuntu 18.04]

## Minimal Working Example (MWE)
Provide a minimal working example that illustrates the bug.
```c++
int main() { 
  // Code here.
  return 0;
}
```

## Steps to Reproduce Bug
1. Do this
2. Do that
3. Compile that

### Expected behaviour
A clear and concise description of what you expected to happen.

### Actual behaviour
Explain the actual behviour observed.

### Screenshots
If applicable, add screenshots to help explain your problem.

rseac, 2023-08-28:
@Elwin-zhao  Have you been able to successfully install the toolchain?  Can you tell me how far you got through the generation?  

We have documentation here: https://systemc-clang.readthedocs.io/en/latest/ if you haven't seen it already.

Elwin-zhao, 2023-08-29:
I have been able to successfully install the toolchain, and I have been able to use scripts/run.sh to generate symbol table of a  systemc program, but how to generate IR?

rseac, 2023-08-30:
You can use the `scripts/run-cpp-sv.sh` script. You will see two files being generated. One of those will have `_hdl.txt`, which contains the HCode. Note that this script will generate both the SystemVerilog and the HCode.  The files will be placed at the same location where the source file exists.

Elwin-zhao, 2023-08-31:
@rseac There is an error which is thrown after I use the scripts/run-cpp-sv.sh 
"raise SystemCClangFatalError('Target file: {} is not generated by systemc-clang binary'.format(t))"

Elwin-zhao, 2023-09-01:
@rseac I'm afraid that invoking systemc-clang directly will not generate _hdl.txt. 
[image]
And I also try to use scripts/run-cpp-sv.sh to generate file, however there is an error:
[image]

Elwin-zhao, 2023-09-01:
@rseac And I have another question, after I generate IR, how can I turn it into executable binary file? As you know, we can write IR and use llvm to compile and execute it.
 **e.g. llvm-as test.ll -o output.bc, lli output.bc.** 
So after I generate IR by using systemc-clang, how I can compile and execute it?

rseac, 2023-09-01:
> @rseac I'm afraid that invoking systemc-clang directly will not generate _hdl.txt. [image] And I also try to use scripts/run-cpp-sv.sh to generate file, however there is an error: [image]

The first screenshot actually shows systemc-clang running.  So that's good. I want to check that you're building systemc-clang with the `-DHDL` flag?  That enables the HDL plugin.  You would provide this when running `cmake`.

> @rseac And I have another question, after I generate IR, how can I turn it into executable binary file? As you know, we can write IR and use llvm to compile and execute it. **e.g. llvm-as test.ll -o [...]

The IR called HCode in this tool is a custom IR. We currently have a backend that targets SystemVerilog.  You would use the `-DHDL` to generate the SystemVerilog too.

Elwin-zhao, 2023-09-15:
@rseac I have enabled DHDL and I use run.sh to generate _hdl.txt. Actually the file _hdl.txt can be created but it is empty. Also I use run-cpp-sv.sh to generate _hdl.txt.v, the file is also empty

mayagokhale, 2023-09-15:
Did you do a cmake with -DHDL=ON? (and delete the CMakeCache.txt in the build directory first?

Elwin-zhao, 2023-10-08:
Suppose that I want to generate 

> The first screenshot actually shows systemc-clang running. So that's good. I want to check that you're building systemc-clang with the `-DHDL` flag? That enables the HDL plugin. You would provide [...]

Suppose I want to generate LLVM-based IR, how should I modify the source code? HCode does not have module information which is important for systemc language, because I want to analyse IR and do some optimization such as conflict detection and handling @rseac @mayagokhale

rseac, 2023-10-09:
@Elwin-zhao Let us address some of these issues one at a time. 

First, let us see if you're compiling with the HDL plugin.

Looking at your first screen capture, I see that you do not have an output to the console that shows `HDL plugin`. This means that you haven't compiled with the `-DHDL=on` flag. You should see an output such as:

```bash
 $  ./systemc-clang                                
HDL plugin
....
```
Do you see that output to the console?  Let us confirm this before we proceed to addressing the other issues.

Elwin-zhao, 2023-10-12:
[image]
I have compiled with the -DHDL=on but there is an error @rseac

rseac, 2023-10-13:
@Elwin-zhao It's good that the `HDL plugin` is showing up. This means that the plugin compiled.

Can you try the command that you see but just remove the `-debug` flag and let me know what happens?

Elwin-zhao, 2023-10-13:
> @Elwin-zhao It's good that the `HDL plugin` is showing up. This means that the plugin compiled.
> 
> Can you try the command that you see but just remove the `-debug` flag and let me know what happens?
[image] @rseac

rseac, 2023-10-13:
@Elwin-zhao  Did you compile clang from source or are you using the pre-built binaries?  I want to try and setup the same build on my side to try and reproduce this error.

Elwin-zhao, 2023-10-16:
> that you're using clang 15.0.0. Could you try with 15.0.6? I have noticed when bumping up versions of clang that there were some surprising issues that got fixed in the clang source code ...
I compiled clang from source.
@rseac

rseac, 2023-10-16:
@Elwin-zhao For now, can you try using the pre-built binaries for 15.0.6 from https://github.com/llvm/llvm-project/releases/tag/llvmorg-15.0.6?

Hiren

Elwin-zhao, 2023-10-19:
Another question. Suppose I want to get the AST generated by Clang, how can I get it? Should I modify the source code? If so, how should I modify it? @rseac

Elwin-zhao, 2023-10-20:
[image]
There is an error when I bumped Clang to 15.0.6. How to fix this bug? @rseac

Elwin-zhao, 2023-10-24:
> @Elwin-zhao For now, can you try using the pre-built binaries for 15.0.6 from https://github.com/llvm/llvm-project/releases/tag/llvmorg-15.0.6?
> 
> Hiren

This problem has been fixed after I using clang 15.0.6 @rseac

closed 2023-10-30 by Elwin-zhao

rseac, 2023-10-31:
Great to hear this @Elwin-zhao