anikau31/systemc-clang #403: How to get the ast and symbol table through this tool ?

Elwin-zhao, 2023-10-30:
# Description
A clear and concise description of what the bug is.

# Reproducing the bug 

## Versions 
- systemc-clang: [i.e. 0.1]
- CXX compiler: [i.e. clang-7.0.0]
- CXX flags: [i.e. llvm-config --cxxflags]
- SystemC: [i.e. 2.3.3]
- Operating System: [i.e. Ubuntu 18.04]

## Minimal Working Example (MWE)
Provide a minimal working example that illustrates the bug.
```c++
int main() { 
  // Code here.
  return 0;
}
```

## Steps to Reproduce Bug
1. Do this
2. Do that
3. Compile that

### Expected behaviour
A clear and concise description of what you expected to happen.

### Actual behaviour
Explain the actual behviour observed.

### Screenshots
If applicable, add screenshots to help explain your problem.

rseac, 2023-10-31:
You wish to get the AST generated by clang?

Elwin-zhao, 2023-11-01:
> You wish to get the AST generated by clang?

Yes, and I wish to get the symbol table too.

Elwin-zhao, 2023-11-03:
Did this project support this functionality?

closed 2023-11-03 by mayagokhale

rseac, 2023-11-05:
@Elwin-zhao systemc-clang gives you access to what clang gives you access too.  This comes from `ASTContext`. Take a look at `SystemCConsumer` class and its `getContext` method.  It gives access to the main context necessary to get the AST from (same approach used if you were to use vanilla clang).  Hope this helps.

Elwin-zhao, 2023-11-09:
Thanks, I have taken a look at SystemCConsumer and getContext method. And which class and method should I take a look at if I want to get the symbol table? @mayagokhale @rseac