anikau31/systemc-clang #404: The docker can not work

megleo, 2023-11-02:
# Description
In the docker， using the command "cmake /systemc-clang -DHDL=ON -DENABLE_TESTS=ON -DENABLE_VERILOG_TESTS=ON -G Ninja && ninja && ctes"
/usr/bin/ld: cannot find -lclangSupport
collect2: error: ld returned 1 exit status

# Reproducing the bug 

## Versions 
https://systemc-clang.readthedocs.io/en/latest/docker.html
- systemc-clang: [i.e. 0.1]
- CXX compiler: [i.e. clang-7.0.0]
- CXX flags: [i.e. llvm-config --cxxflags]
- SystemC: [i.e. 2.3.3]
- Operating System: [i.e. Ubuntu 18.04]

## Minimal Working Example (MWE)
Provide a minimal working example that illustrates the bug.
```c++
int main() { 
  // Code here.
  return 0;
}
```

## Steps to Reproduce Bug
1. Do this
2. Do that
3. Compile that

### Expected behaviour
A clear and concise description of what you expected to happen.

### Actual behaviour
Explain the actual behviour observed.

### Screenshots
If applicable, add screenshots to help explain your problem.

megleo, 2023-11-03:
Hi, I follow https://systemc-clang.readthedocs.io/en/latest/docker.html
"cmake /systemc-clang -DHDL=ON -DENABLE_TESTS=ON -DENABLE_VERILOG_TESTS=ON -G Ninja && ninja && ctest"

/usr/bin/ld: cannot find -lclangSupport

megleo, 2023-11-03:
您好，解决了， 是我的system-clang的版本和提供的docker的版本没对应起来，导致的。 我理解主要llvm
没对应起来，

megleo, 2023-11-03:
resolved

closed 2023-11-03 by megleo

mayagokhale, 2023-11-03:
In English: "Hello, the problem is solved. The problem is that my system-clang version does not match the provided docker version. I understand main llvm
It doesn't correspond."