anikau31/systemc-clang #405: Which class should I pay attention to if I want to get the symbol table

Elwin-zhao, 2023-11-13:
# Description
A clear and concise description of what the bug is.

# Reproducing the bug 

## Versions 
- systemc-clang: [i.e. 0.1]
- CXX compiler: [i.e. clang-7.0.0]
- CXX flags: [i.e. llvm-config --cxxflags]
- SystemC: [i.e. 2.3.3]
- Operating System: [i.e. Ubuntu 18.04]

## Minimal Working Example (MWE)
Provide a minimal working example that illustrates the bug.
```c++
int main() { 
  // Code here.
  return 0;
}
```

## Steps to Reproduce Bug
1. Do this
2. Do that
3. Compile that

### Expected behaviour
A clear and concise description of what you expected to happen.

### Actual behaviour
Explain the actual behviour observed.

### Screenshots
If applicable, add screenshots to help explain your problem.

Elwin-zhao, 2023-11-13:
I have taken a look at SystemCConsumer and getContext method. And which class and method should I take a look at if I want to get the symbol table? @mayagokhale @rseac

rseac, 2023-11-14:
@Elwin-zhao  I'm not entirely sure what you're looking for from within systemc-clang.  Could you give me a concrete example of what you are looking for when asking about the symbol table?

Elwin-zhao, 2023-11-15:
I want to use symbol table to

> @Elwin-zhao I'm not entirely sure what you're looking for from within systemc-clang. Could you give me a concrete example of what you are looking for when asking about the symbol table?

I want to get the value of the variable at every moment during runtime, because it involves multi-threaded reading and writing of variables. By extracting the information of the abstract syntax tree and putting it into the symbol table, the value of the runtime variable at each moment can be obtained when the systemc simulation program is running, which can be used to detect data conflicts. @rseac

rseac, 2023-11-18:
Clang provides static information. As far as I know, it does not provide runtime information.  You can access the AST.  This is what I had mentioned earlier where you can get Clang's AST from the ASTContext. https://clang.llvm.org/docs/IntroductionToTheClangAST.html 

Hope this helps.

closed 2023-11-19 by rseac